konnected-io/konnected-security #45: Fails, reboots when taking input, at the 'application' step

jaredbehunin, 2017-11-13:
Hello, 
Great project Nate.  Sharing my issue here to see if there is a solution among the masses.   Non-technical, so I need help from others out there who understand this stuff.  I have two devices behaving identically.  I would think there is a solution of some sort...

**My setup:**  
Modem: Xfinity rented. TG862G rev5, software: 9.1.103 
Wifi: Luma Surround WiFi, with three hubs.  latest firmware 2017.10.25-0
NodeMCU ESP8266 12, flashed with Konnected Security 2.0.5
build 20170825 powered by Lua 5.1.4 on SDK 1.5.4.1(39cb9a32)

**The behavior:**
The setup is fine from plug in to setting up pins.  Using an open/close sensor on any pin D1, D2, D5, D6, D7, when the first change in state is detected (0 or 1), the device errors and resets 5-7 times. getting stuck at 'application'.  Eventually it stabilizes.  But, the moment another change is detected, it cycles again.  I've flashed it and reinstalled, same behavior.

**The problem:**
its getting stuck in the application step.  I believe the error is happening at 
line 14: timeout:register(10000, tmr.ALARM_SEMI, node.restart)
I think it is stalling out after 10 seconds and forcing the restart, then if I had to guess, it takes the http a portion of time to jettison whatever caused it to time out, then once its gone, it settles down again...

Please help.  This project is awesome and I would really like to get it working.

**<begin copy/paste>** 
Heap:	31904	HTTP Call:	204	Pin:	1	State:	0
HTTP client: Disconnected with error: -11
HTTP client: Connection timeout

 ets Jan  8 2013,rst cause:2, boot mode:(3,6)

load 0x40100000, len 24580, room 16 
tail 4
chksum 0xa2
load 0x3ffe8000, len 2268, room 4 
tail 8
chksum 0x7d
load 0x3ffe88dc, len 136, room 0 
tail 8
chksum 0x89
csum 0x89
„ãìÃ'ä’{Ãón<ä$dld`„ã;›$ŒdŒþ

NodeMCU 1.5.4.1
Konnected Security 2.0.5
 build 20170825 powered by Lua 5.1.4 on SDK 1.5.4.1(39cb9a32)
Heap: 	41096	Initializing Konnected
Heap: 	36664	Version: 	2.1.0
Heap: 	35016	Connecting to Wifi..
> Heap: 	39304	Wifi connected with IP: 	192.168.55.249	255.255.255.0	192.168.55.1
Heap: 	35656	UPnP: 	Listening for UPnP discovery
Heap: 	35240	HTTP: 	Starting server at http://192.168.55.249:17068
Heap: 	34896	Loaded: 	server
Heap:	32048	Initializing sensor pin:	1
Heap: 	31792	Loaded: 	application
Heap:	31904	HTTP Call:	204	Pin:	1	State:	1

jaredbehunin, 2017-11-13:
update:  I was able to get the system to work with the actuator devices.  It's the sensors (0,1) that is contributing to the problem. When no 1/0 sensors are configured, and only actuator type devices are enabled, the system works and sends signals instantaneously.

jaredbehunin, 2017-11-16:
Nate, any update/comments before I send my devices back for a refund?

vseven, 2017-11-16:
Not to get too far into this but hace you tried just reflashing the firmware?  I know when I have had an issue in the past this is the 1st thing I did and it always seemed to fix the issue.

jaredbehunin, 2017-11-17:
I tried refreshing the firmware per these instructions... https://docs.konnected.io/security-alarm-system/upgrading

vseven, 2017-11-17:
I had one board that was giving me some weird trouble so what I did was, from that page, the "Flashing Firmware: Windows Users" to make sure the firmware was good.  I then used ESPlorer to connect to the device and I deleted every file off of it.  Once everything was deleted I followed the "Manual Software Upload via USB" instructions to put the latest version back on.  Rebooted and it has worked fine ever since.  If it still doesn't work I'd say its a bad board but the fact its talking at all would lead me to believe there is just something messed up somewhere on it.

heythisisnate, 2018-07-25:
This problem is resolved in the latest Konnected firmware 2.2.0+

closed 2018-07-25 by heythisisnate