konnected-io/konnected-security #49: Failure to upload after flashing

rdmorin, 2017-12-01:
This worked seamlessly on my first attempt. The second ESP8266 NodeMCU module I tried has worked with other firmware. Every time I try to flash it with Konnected firmware it fails to reboot. The LED is flashing rapidly as if it is rebooting and not finding the bootloader. Is there a way around this? If I re-flash with the Arduino API it seems to work still, so the chip does not seem to have any permanent damage. 

./scripts/flash
esptool.py v2.2-dev
Connecting....
Detecting chip type... ESP8266
Chip is ESP8266EX
Uploading stub...
Running stub...
Stub running...
Configuring flash size...
Auto-detected Flash size: 4MB
Flash params set to 0x0240
Compressed 479444 bytes to 311295...
Wrote 479444 bytes (311295 compressed) at 0x00000000 in 27.5 seconds (effective 139.6 kbit/s)...
Hash of data verified.

Leaving...
Hard resetting...
opening port /dev/tty.SLAB_USBtoUART with 115200 baud
Traceback (most recent call last):
  File "//anaconda/bin/nodemcu-uploader", line 11, in <module>
    sys.exit(main_func())
  File "//anaconda/lib/python2.7/site-packages/nodemcu_uploader/main.py", line 252, in main_func
    uploader = Uploader(args.port, args.baud, start_baud=args.start_baud)
  File "//anaconda/lib/python2.7/site-packages/nodemcu_uploader/uploader.py", line 78, in __init__
    __sync()
  File "//anaconda/lib/python2.7/site-packages/nodemcu_uploader/uploader.py", line 76, in __sync
    raise DeviceNotFoundException('Device not found or wrong port')
nodemcu_uploader.exceptions.DeviceNotFoundException: Device not found or wrong port

heythisisnate, 2017-12-01:
It sometimes takes a minute or two to reboot. It looks like you're using my `flash` script, which is designed to wait and retry after a couple seconds for it to reboot after flashing. Can you try giving it a minute and see if it eventually does upload successfully?

rdmorin, 2017-12-01:
I noticed it was in a loop but I waited ages to no avail. After some searching I decided the flash was failing and found that a complete erase of the flash and fresh run of the script solved the issue. 
Thanks for your help and for the great software!

closed 2017-12-01 by heythisisnate