konnected-io/konnected-security #51: HTTP client: Disconnected with error: -11

dmunson424, 2017-12-18:
You have a great product and code Nate.  As I am attempting to get my first NodeMCU online, the unit connects to Wifi, loads the server, initializes the six sensor pins, loads the application, and then disconnects with a HTTP client error.  The printout is shown below.  Any advice?  

NodeMCU 1.5.4.1
Konnected Security 2.0.5
 build 20170825 powered by Lua 5.1.4 on SDK 1.5.4.1(39cb9a32)
Heap: 	41264	Initializing Konnected
Heap: 	36776	Version: 	2.1.0
Heap: 	35024	Connecting to Wifi..
> Heap: 	39648	Wifi connected with IP: 	192.168.0.22	255.255.255.0	192.168.0.1
Heap: 	36120	UPnP: 	Listening for UPnP discovery
Heap: 	35688	HTTP: 	Starting server at http://192.168.0.22:9827
Heap: 	35344	Loaded: 	server
Heap:	31656	Initializing sensor pin:	5
Heap:	31608	Initializing sensor pin:	7
Heap:	31608	Initializing sensor pin:	2
Heap:	31608	Initializing sensor pin:	1
Heap:	31608	Initializing sensor pin:	9
Heap:	31608	Initializing sensor pin:	6
Heap: 	31360	Loaded: 	application
HTTP client: Disconnected with error: -11
HTTP client: Connection timeout
HTTP client: Connection timeout

 ets Jan  8 2013,rst cause:2, boot mode:(3,6)

load 0x40100000, len 24580, room 16 
tail 4
chksum 0xa2
load 0x3ffe8000, len 2268, room 4 
tail 8
chksum 0x7d
load 0x3ffe88dc, len 136, room 0 
tail 8
chksum 0x89
csum 0x89
„âì2‚nì’sƒòN|ì$l$l`Œãr’lŒ$Œþ

heythisisnate, 2017-12-18:
This is a duplicate of #45 

I've seen this issue before a handful of times. We don't know the _exact_ cause but it's almost always related to a leased cable modem from your cable company. Upgrading the firmware or replacing the cable modem is the only workaround right now.

closed 2017-12-18 by heythisisnate

dmunson424, 2017-12-21:
Nate,Just wanted to let you know that your diagnosis is right.  We changed out the cable company's modem/router, and now everything is working like a champ.Thanks a bunch.Dan