konnected-io/konnected-security #53: Home assistant

rickevans86, 2018-01-18:
Wondering when this was getting home assistant support as Samsung smart things is unavailable in my country

heythisisnate, 2018-01-18:
I'm working on finishing up some documentation for the first beta release. Should be an announcement by the end of this weekend.

rickevans86, 2018-01-18:
Thanks for the fast reply
Can't wait to test it out

lostage, 2018-01-27:
If you need some beta testers I'm sure there are many of us that would help. Or did I miss the announcement (totally possible)?

heythisisnate, 2018-01-27:
I'm of course behind on testing and writing up documentation. There a short write-up here if you want to try and play with the early proof of concept: https://github.com/konnected-io/home-assistant/tree/custom_component/konnected

pepowdylan, 2018-02-23:
Nate,

I'm getting a few errors with your early proof of concept. Can you help out?

Log Details (ERROR)
Thu Feb 22 2018 21:05:42 GMT-0500 (EST)

Error handling request
Traceback (most recent call last):
  File "/usr/lib/python3.6/site-packages/aiohttp/web_protocol.py", line 272, in data_received
    messages, upgraded, tail = self._request_parser.feed_data(data)
  File "aiohttp\_http_parser.pyx", line 295, in aiohttp._http_parser.HttpParser.feed_data
aiohttp.http_exceptions.BadHttpMessage: 400, message='invalid character in header'

heythisisnate, 2018-06-13:
Konnected support is built in to Home Assistant 0.70 and newer.
https://www.home-assistant.io/components/konnected

closed 2018-06-13 by heythisisnate