konnected-io/konnected-security #54: Smartapp not able to find esp8266

gmbuchanan, 2018-01-21:
I am up to the step where I would ordinarily mount my ESP8266 and start wiring. However, the connection between the Konnected smartapp and the esp8266 is having issues. 

When I look at the smartapps on my phone, there is no app for Konnected. When I try to add it manuallyvia "automation", "SmartApps", "add a SmartApp", "My Apps" - I see Konnected. When I tap on this i see an option to "Add a Konnected device". When I tap on this, I see, via esplorer console, "HTTP: Discovery" and then "Responded to uPnP Discovery request" -- but the smartthing app on the iphone just spins -- eventually have to force close the app. 

I can view info on esp8266 web interface, connection via wifi is 100%. 

Web interface also shows:
Software -> 2.1.0
Firmware -> 2.0.5
IP Address and gateway are consistent with my network. 

any suggestions or areas to look at ? 

Thank you.

sreknob, 2018-05-02:
I'm having the exact same issue.

No device ever found under "Add a Konnected device" in ST App.

I've checked with wireshark and I'm getting discovery packets from the ESP on port 1900 as expected on my computer.

I have restarted by my STHub v2 and ESP.

ESPlorer console showing discovery request responses as below:
Heap: 	27888	Responded to UPnP Discovery request
Heap: 	28008	Responded to UPnP Discovery request
Heap: 	24976	HTTP: 	Discovery
Heap: 	27824	Responded to UPnP Discovery request
Heap: 	27816	Responded to UPnP Discovery request
Heap: 	27408	HTTP: 	Status
Heap: 	25952	Responded to UPnP Discovery request
Heap: 	26072	Responded to UPnP Discovery request
Heap: 	25880	Responded to UPnP Discovery request

I'm using Ubiquiti Unifi gear and everything else with my network seems in order.

Any thoughts?

heythisisnate, 2018-05-03:
@sreknob this log looks like it's working properly.
Try removing and re-adding the _Konnected (Connect)_ SmartApp to reset the internal state stored on SmartThings.

sreknob, 2018-05-11:
Hi Nate -
Thanks for the suggestion, but no dice. I've rebooted everything multiple times and I have been sniffing my UPnP and SSDP traffic. I have lots of Sonos and chromecasts, etc that all use SSDP without issue. I had given up but saw there was a new release out, so trying again.
I updated to the latest 2.2.0 release from a couple days ago and I see that the logging has changed to include the IP address it is responding to -- great idea!
I see both the SSDP packet being sent in the ST IDE and also see my ESP responding in live logging in ESPlorer.
I have verified that I'm connected to regional IDE https://graph-na04-useast2.api.smartthings.com/ide/logs

SOOOOOOO..... I'm going through your troubleshooting steps (looks like recently updated) as well as the install instructions. I just removed and reinstalled the smart app in the IDE from Github... looking back it seems that I had OAuth enabled on the wrong smart app, so that was probably my issue and completely my fault... Just posting here in case it might help someone else... I have now found my device! Yay!!!

heythisisnate, 2018-05-11:
Great. Glad you got it working. For future searchers finding this issue, there are some Discovery troubleshooting steps here: https://help.konnected.io/support/solutions/articles/32000023644-device-discovery-troubleshooting

closed 2018-05-11 by heythisisnate