konnected-io/konnected-security #59: PANIC: unprotected error in call to Lua API (?:0: attempt to concatenate a nil value)

sreknob, 2018-05-23:
Hi -
I set up two ESPs running Konnected and have been getting random reboots on both of them. I first noticed as the power to one of my relays kept getting turned off and my relay switches were out of sync with Home Assistant. I confirmed this by loading up the Konnected Device status webpage on both of them and noted that with a refresh the uptime would decrease with a refresh every once and a while. I pulled down one of the controllers to hook it up to serial and I caught one of the reboots. This is the error reported. I have no sensors hooked up. Googling suggest this is a missing HTTP module in the firmware but I see it listed as included in "firmware/2.2.0/app/include/user_modules.h"
Thoughts?

```
Heap: 	32936	Responded to UPnP Discovery request from 192.168.20.5:56672
Heap: 	32952	Responded to UPnP Discovery request from 192.168.20.26:43810
PANIC: unprotected error in call to Lua API (?:0: attempt to concatenate a nil value)

 ets Jan  8 2013,rst cause:2, boot mode:(3,6)

load 0x40100000, len 26212, room 16 
tail 4
chksum 0xfc
load 0x3ffe8000, len 2340, room 4 
tail 0
chksum 0xeb
load 0x3ffe8924, len 136, room 8 
tail 0
chksum 0x2b
csum 0x2b

Konnected firmware 2.2
NodeMCU 2.2.0.0 build 20180405 powered by Lua 5.1.4 on SDK 2.2.1(cfd48f3)
Heap: 	39856	Initializing Konnected
Heap: 	35312	Version: 	2.2.0
Heap: 	33688	Connecting to Wifi..
> Heap: 	37536	Wifi connected with IP: 	192.168.20.199	255.255.255.0	192.168.20.1
Heap: 	37504	UPnP: 	Listening for UPnP discovery
Heap: 	37296	HTTP: 	Starting server at http://192.168.20.199:9664
Heap: 	37128	Loaded: 	server
Heap:	32160	Initializing actuator pin:	1	Trigger:	0
Heap:	32128	Initializing actuator pin:	2	Trigger:	0
Heap:	32128	Initializing actuator pin:	5	Trigger:	0
Heap:	32128	Initializing actuator pin:	6	Trigger:	0
Heap: 	31880	Loaded: 	application
Heap: 	32936	Responded to UPnP Discovery request from 192.168.20.5:56672
Heap: 	26992	HTTP: 	Discovery
```

heythisisnate, 2018-05-23:
Thanks for posting this. I haven't seen this particular issue before. If you let this run a few more times, does it always get this error after responding to `192.168.20.26:43810`?

sreknob, 2018-05-24:
A couple more panics overnight.

```
Heap: 	32640	Responded to UPnP Discovery request from 192.168.20.5:56672
Heap: 	32640	Responded to UPnP Discovery request from 192.168.20.17:37931
Heap: 	32640	Responded to UPnP Discovery request from 192.168.20.5:56672
Heap: 	32640	Responded to UPnP Discovery request from 192.168.20.14:34366
Heap: 	32632	Responded to UPnP Discovery request from 192.168.20.16:34036
Heap: 	32640	Responded to UPnP Discovery request from 192.168.20.21:35149
Heap: 	32640	Responded to UPnP Discovery request from 192.168.20.23:49972
Heap: 	32640	Responded to UPnP Discovery request from 192.168.20.5:56672
Heap: 	32640	Responded to UPnP Discovery request from 192.168.20.5:56672
PANIC: unprotected error in call to Lua API (?:0: attempt to concatenate a nil value)
```

```
Heap: 	32640	Responded to UPnP Discovery request from 192.168.20.21:35149
Heap: 	32648	Responded to UPnP Discovery request from 192.168.20.23:49972
Heap: 	32640	Responded to UPnP Discovery request from 192.168.20.5:56672
Heap: 	32472	Responded to UPnP Discovery request from 192.168.20.192:35296
Heap: 	32648	Responded to UPnP Discovery request from 192.168.20.26:37989
Heap: 	32648	Responded to UPnP Discovery request from 192.168.20.5:56672
Heap: 	32584	Responded to UPnP Discovery request from 192.168.20.5:56672
Heap: 	32640	Responded to UPnP Discovery request from 192.168.20.22:40940
Heap: 	32640	Responded to UPnP Discovery request from 192.168.20.18:39892
Heap: 	32640	Responded to UPnP Discovery request from 192.168.20.5:56672
Heap: 	32640	Responded to UPnP Discovery request from 192.168.20.17:37931
Heap: 	32640	Responded to UPnP Discovery request from 192.168.20.5:56672
PANIC: unprotected error in call to Lua API (?:0: attempt to concatenate a nil value)
```
My Google homes are 14, 16, 17, 18, 21, 22, 23, 26, having request shortly before the panic. Perhaps just co-incidence?

Last discovery request is from my Windows 10 desktop in both before the panic. Are there additional ways to help debug this?

Thanks!

sreknob, 2018-05-25:
Continuing to follow this. Four more reboots with panics. Not certain it has to do with specific discovery packets as 2 have happened after discovery from Plex Server and two have happened after discovery from Google Homes. Plex and Google Homes make up the bulk of my SSDP traffic on my network. Other things are Nest, Ecobee, some motorola and d-link IP cameras, Home Assistant, Sonoff-Tasmota, SmartThings Hub, Wink Hub2, Hue Hub 2, WeMo devices, 8 Chromecast video and audio devices, multiple Sonos units, Apple TV 3s, Amazon FireTV, plus my UniFi gear.... I guess I kinda have a lot of SSDP traffic :-/
Any tips on where to look for additional debugging? I’m currently running a wireshark capture of SSDP packets in case it shows something funky. Any other ideas?

heythisisnate, 2018-05-25:
Can I ask you to add some debugging `print` statements in a few places. It's hard to tell where the error is coming from via this error message.
I suspect that it's coming from `ssdp_response.lua`. Could you add the line `print(d)` between lines 7 and 8 in this file and upload it to your device?

sreknob, 2018-05-25:
Done.
Will post an updated log with the next error.
Thanks!

sreknob, 2018-05-26:
```
Heap: 	32808	Responded to UPnP Discovery request from 192.168.20.26:41837
M-SEARCH * HTTP/1.1
Host: 239.255.255.250:1900
Man: "ssdp:discover"
ST: ssdp:all
MX: 5

Heap: 	32816	Responded to UPnP Discovery request from 192.168.20.5:54225
M-SEARCH * HTTP/1.1
HOST: 239.255.255.250:1900
MAN: "ssdp:discover"
MX: 2
ST: ssdp:all

Heap: 	32816	Responded to UPnP Discovery request from 192.168.20.17:42666
M-SEARCH * HTTP/1.1
Host: 239.255.255.250:1900
Man: "ssdp:discover"
ST: ssdp:all
MX: 5

Heap: 	32824	Responded to UPnP Discovery request from 192.168.20.5:54225
M-SEARCH * HTTP/1.1
HOST: 239.255.255.250:1900
MAN: "ssdp:discover"
MX: 2
ST: ssdp:all

Heap: 	32784	Responded to UPnP Discovery request from 192.168.20.16:47887
M-SEARCH * HTTP/1.1
HOST: 239.255.255.250:1900
MAN: "ssdp:discover"
MX: 2
ST: ssdp:all

PANIC: unprotected error in call to Lua API (?:0: attempt to concatenate a nil value)

 ets Jan  8 2013,rst cause:2, boot mode:(3,7)
```

sreknob, 2018-05-28:
Unfortunately, I don't think that additional print statement has a lot more info...

```
Heap: 	32616	Responded to UPnP Discovery request from 192.168.20.18:33319
M-SEARCH * HTTP/1.1
HOST: 239.255.255.250:1900
MAN: "ssdp:discover"
MX: 2
ST: ssdp:all

Heap: 	32616	Responded to UPnP Discovery request from 192.168.20.17:46329
M-SEARCH * HTTP/1.1
HOST: 239.255.255.250:1900
MAN: "ssdp:discover"
MX: 2
ST: ssdp:all

Heap: 	32616	Responded to UPnP Discovery request from 192.168.20.16:46338
M-SEARCH * HTTP/1.1
HOST: 239.255.255.250:1900
MAN: "ssdp:discover"
MX: 2
ST: ssdp:all

Heap: 	32600	Responded to UPnP Discovery request from 192.168.20.21:40936
M-SEARCH * HTTP/1.1
Host: 239.255.255.250:1900
Man: "ssdp:discover"
ST: ssdp:all
MX: 5

Heap: 	32616	Responded to UPnP Discovery request from 192.168.20.5:54225
M-SEARCH * HTTP/1.1
HOST: 239.255.255.250:1900
MAN: "ssdp:discover"
MX: 2
ST: ssdp:all

PANIC: unprotected error in call to Lua API (?:0: attempt to concatenate a nil value)

 ets Jan  8 2013,rst cause:2, boot mode:(3,7)

load 0x40100000, len 26212, room 16 
tail 4
chksum 0xfc
load 0x3ffe8000, len 2340, room 4 
tail 0
chksum 0xeb
load 0x3ffe8924, len 136, room 8 
tail 0
chksum 0x2b
csum 0x2b
Œãäƒoì›rƒûg|ìdldl`b“Ÿ|{’lÜoàƒoã

Konnected firmware 2.2
NodeMCU 2.2.0.0 build 20180405 powered by Lua 5.1.4 on SDK 2.2.1(cfd48f3)
Heap: 	39856	Initializing Konnected
Heap: 	35304	Version: 	2.2.0
Heap: 	33600	Connecting to Wifi..
> Heap: 	37536	Wifi connected with IP: 	192.168.20.199	255.255.255.0	192.168.20.1
Heap: 	37504	UPnP: 	Listening for UPnP discovery
Heap: 	37296	HTTP: 	Starting server at http://192.168.20.199:9664
Heap: 	37128	Loaded: 	server
Heap:	32136	Initializing sensor pin:	7
Heap:	32104	Initializing actuator pin:	1	Trigger:	0
Heap:	32104	Initializing actuator pin:	2	Trigger:	0
Heap:	32104	Initializing actuator pin:	5	Trigger:	0
Heap:	32104	Initializing actuator pin:	6	Trigger:	0
Heap: 	31832	Loaded: 	application
Heap:	35680	HTTP Call:	200	state	0	pin	7
M-SEARCH * HTTP/1.1
Host: 239.255.255.250:1900
Man: "ssdp:discover"
ST: ssdp:all
MX: 5

```

heythisisnate, 2019-01-17:
I _think_ this should be resolved in the latest 2.2.5 firmware. Can you confirm?

closed 2019-01-17 by heythisisnate

sreknob, 2019-01-18:
Thanks Nate, I’ll load it up tonight and let you know!

sreknob, 2019-01-19:
My test device stayed up while my others rebooted, so looks like it’s fixed! Thanks!