konnected-io/konnected-security #60: Reboot loop

sreknob, 2018-05-23:
So, I just updated my config to do some additional testing and now I'm stuck in a reboot loop as soon as my ESP gets the config information from Home Assistant. Each time it errors with HTTP Call: -1 state 1 pin 7
I'm running a docker of the latest dev branch of Home Assistant (0.70.0.dev0) which has konnected integration.
I had a jumper on D7 to ground during the update but it continues the loop even with nothing connected.
Log shows:
```
Konnected firmware 2.2
NodeMCU 2.2.0.0 build 20180405 powered by Lua 5.1.4 on SDK 2.2.1(cfd48f3)
Heap: 	39856	Initializing Konnected
Heap: 	35312	Version: 	2.2.0
Heap: 	33688	Connecting to Wifi..
> print(uart.setup(0, 15200, 8, 0, 1, 1 ))
115200
> 
Communication with MCU..Got answer! Communication with MCU established.
AutoDetect firmware...

Can't autodetect firmware, because proper answer not received (may be unknown firmware). 
Please, reset module or continue.
 IP: 	192.168.20.199	255.255.255.0	192.168.20.1
Heap: 	37824	UPnP: 	Listening for UPnP discovery
Heap: 	37608	HTTP: 	Starting server at http://192.168.20.199:9664
Heap: 	37432	Loaded: 	server
Heap:	32024	Initializing actuator pin:	1	Trigger:	0
Heap:	31992	Initializing actuator pin:	2	Trigger:	0
Heap:	31992	Initializing actuator pin:	5	Trigger:	0
Heap:	31992	Initializing actuator pin:	6	Trigger:	0
Heap: 	31736	Loaded: 	application
Heap: 	30080	HTTP: 	Status
Heap: 	28304	HTTP: 	Settings
Heap: 	25008	Wrote: 	smartthings.lua
Heap: 	24976	Wrote: 	sensors.lua
Heap: 	24704	Wrote: 	actuators.lua
Settings updated! Restarting in 5 seconds...

 ets Jan  8 2013,rst cause:2, boot mode:(3,6)

load 0x40100000, len 26212, room 16 
tail 4
chksum 0xfc
load 0x3ffe8000, len 2340, room 4 
tail 0
chksum 0xeb
load 0x3ffe8924, len 136, room 8 
tail 0
chksum 0x2b
csum 0x2b

Konnected firmware 2.2
NodeMCU 2.2.0.0 build 20180405 powered by Lua 5.1.4 on SDK 2.2.1(cfd48f3)
Heap: 	39856	Initializing Konnected
Heap: 	35312	Version: 	2.2.0
Heap: 	33688	Connecting to Wifi..
> Heap: 	37536	Wifi connected with IP: 	192.168.20.199	255.255.255.0	192.168.20.1
Heap: 	37504	UPnP: 	Listening for UPnP discovery
Heap: 	37288	HTTP: 	Starting server at http://192.168.20.199:9664
Heap: 	37112	Loaded: 	server
Heap:	32128	Initializing sensor pin:	7
Heap:	32096	Initializing actuator pin:	1	Trigger:	0
Heap:	32096	Initializing actuator pin:	2	Trigger:	0
Heap:	32096	Initializing actuator pin:	5	Trigger:	0
Heap:	32096	Initializing actuator pin:	6	Trigger:	0
Heap: 	31848	Loaded: 	application
Heap: 	32488	Responded to UPnP Discovery request from 192.168.20.18:38442
Heap: 	31968	Responded to UPnP Discovery request from 192.168.20.5:56672
Heap: 	25528	HTTP: 	Discovery
HTTP client: Connection timeout
Heap:	35248	HTTP Call:	-1	state	1	pin	7

 ets Jan  8 2013,rst cause:2, boot mode:(3,6)

load 0x40100000, len 26212, room 16 
tail 4
chksum 0xfc
load 0x3ffe8000, len 2340, room 4 
tail 0
chksum 0xeb
load 0x3ffe8924, len 136, room 8 
tail 0
chksum 0x2b
csum 0x2b
```
My config for this unit is:
```
  # Test Module
  - id: 1964f1
    binary_sensors:
      - pin: 7
        type: door
        name: 'Konnected Test Door'
    switches:
      - pin: 1
        name: 'Konnected Test Relay 1'
        activation: low
      - pin: 2
        name: 'Konnected Test Relay 2'
        activation: low
      - pin: 5
        name: 'Konnected Test Relay 3'
        activation: low
      - pin: 6
        name: 'Konnected Test Relay 4'
        activation: low

```

sreknob, 2018-05-23:
FYI, I've re-flashed and tried this on two modules with a fresh re-load of the Konnected firmware and file system with the same result. As soon as it updates the configuration on the board it goes into a reboot loop.
I moved from the dev docker branch to rc (0.70.0b2) also with the same result.
The board that has not reloaded a config doesn't appear to do this. It got it's config during initial boot on a the first dev branch that supported konnected.
Is there some way to check the config that is on the ESP?
The device status page looks appropriate:

```
Sensors[{"state":1,"pin":7}]
--
Actuators[{"trigger":0,"pin":1},{"trigger":0,"pin":2},{"trigger":0,"pin":5},{"trigger":0,"pin":6}]

```
It also doesn't seem to matter what state pin 7 is in, get HTTP Call: -1 with a reboot regardless of 0 or 1.

```
eap: 	31888	Responded to UPnP Discovery request from 192.168.20.21:39440
Heap: 	32480	Responded to UPnP Discovery request from 192.168.20.5:56672
Heap: 	25936	HTTP: 	Discovery
HTTP client: Connection timeout
Heap:	36312	HTTP Call:	-1	state	0	pin	7

 ets Jan  8 2013,rst cause:2, boot mode:(3,6)

load 0x40100000, len 26212, room 16 
tail 4
chksum 0xfc
load 0x3ffe8000, len 2340, room 4 
tail 0
chksum 0xeb
load 0x3ffe8924, len 136, room 8 
tail 0
chksum 0x2b
csum 0x2b

```

I will see if I can reproduce on another branch or pull in the konnected branch of HA instead.

heythisisnate, 2018-05-23:
It looks like it's getting a HTTP timeout, so that's what's causing it to reboot. This is actually expected behavior. It's designed to reboot after a HTTP timeout as a failsafe. For example, if your internet goes it'll constantly reboot (and try to re-send initial state) until internet is back up.
I'd suggest looking at the home assistant server and make sure that it's running. Did you change the hostname or IP of Hass? you can check the device's stored endpoint by executing this on the lua console:
```
=print(require("smartthings").apiUrl)
```

sreknob, 2018-05-24:
Ahhhhh. Good call! That's put me on the right track :-)

The IP address hadn't changed, but I forgot that in between setting up the first boards and my last one that I did set up my Home Assistant server for remote access.

Checking as you advised:
```
=print(require("smartthings").apiUrl)
http://subdomain.domain.tld:8123/api/konnected
```

The issue is that my HA is set up for HTTPS access using a subdomain and an Nginx reverse proxy.
Trying to access it via the subdomain redirects traffic to HTTPS and forwards the port to the local instance at port 8123.

Turns out I had accidentally left the port number in the base_url section under the HTTP configuration. Fixed the reboot loop issue by removing the port from my base_url parameter.

HOWEVER, it appears that Konnected is not happily handling the 301 redirect of http -> https but spits out the following SSL error that it looks like you're aware of ([NodeMCU problem?](https://github.com/nodemcu/nodemcu-firmware/issues/1707)).

```
E:M 528
HTTP client: Disconnected with error: 46
HTTP client: Connection timeout
```
and whenever the state of a sensor changes I get this and HA does not get the update.

```
HTTP Call:	-1	state	0	pin	7
E:M 1048
HTTP client: Disconnected with error: 46
HTTP client: Connection timeout

```

With the address it's trying to connect to

```
=print(require("smartthings").apiUrl)
http://subdomain.domain.tld/api/konnected
```

I got things working after re-flashing the file system and letting it discover the new config after I had entirely removed base_url from my HA configuration. The module now outputs the IP and is happy.

```
=print(require("smartthings").apiUrl)
http://192.168.20.10:8123/api/konnected
```

I'm not sure if this would also be an issue with SSL if set-up directly in HA (not using a reverse proxy, as in my case). Does there need to be some sort of SSL warning in the Konnected HA documents regarding this configuration or does it work while not using a reverse proxy?

Anyway, thanks for the help!

sreknob, 2018-06-11:
Looks like you addressed the [base_url in a PR](https://github.com/home-assistant/home-assistant/pull/14896) already - amazing! It will hopefully keep others from the same pitfall as me and help partially solve the SSL issues if you use a reverse proxy and no local SSL.
I'll close this issue as it looks like the SSL problem is being addressed elsewhere ([here](https://help.konnected.io/support/discussions/topics/32000001193) and [here](https://community.home-assistant.io/t/konnected-alarm-panel-connect-a-wired-alarm-system-to-ha-new-in-0-70/53620)).

closed 2018-06-11 by sreknob