konnected-io/konnected-security #61: Temperature & Humidity Enhancement Request

Tx-ST-Home, 2018-06-06:
I've purchased 5 AM2302 DHT22 Digital Temperature & Humidty Sensor Module (5V) and updated to Konnected.IO Firmware release 2.2.0. I've configured two DIY Konnected modules with a DHT22 sensor and both are providing live temperature and humidity feedback via the SmartThings app. 

DHT22 - Device Calibration Offset Feature: 
Given the fidelity limits of these relatively inexpensive sensors limits I've found that the two DHT22 report slightly different temp and humidity readings even though both are co-located. Would it be possible to add a calibration feature to the device handler allowing a user to offset the temp and humidity of each of the sensors. I've seen this feature in some of the SmartThings Multipurpose devices so I know it possible. I'm not a groovy coder but would be willing to attempt adding a feature to the handle if I could get some suggestions on where to start. 

Missing DHT Pin Definition: 
I used  the Konnected (Connect) SmartApp to configured two devices on my module P5-Konnected-827777. Pin D1 as a Beep/Blink Switch and Pin D2 as a Temperature & Humidity sensor. However, when I review the P5 module device status the Configuration only shows the Pin D1 entry as an actuator and no entry for the DTH22 on Pin D2 in the Sensors (see attached files). I believe this maybe the reason I can't find this DHT22 sensor in my SmartThing list when using ActionTiles. Any thoughts of suggestions?

Screen Snapshot of Konnect Device Status: 
[Konnect DIY Device Status.pdf](https://github.com/konnected-io/konnected-security/files/2077121/Konnect.DIY.Device.Status.pdf)
[Pin D2 Temperature & Humidity Sensor Assignment & Device Name.pdf](https://github.com/konnected-io/konnected-security/files/2077123/Pin.D2.Temperature.Humidity.Sensor.Assignment.Device.Name.pdf)

-Greg

vseven, 2018-06-06:
The offset is usually done in the DTH:

https://github.com/konnected-io/konnected-security/blob/master/devicetypes/konnected-io/konnected-temperature-humidity-sensor-dht.src/konnected-temperature-humidity-sensor-dht.groovy

You'd modify the "humidity" and "temperature" before it is assigned to the device, lines 69 and 74 from what I can tell.  Here is one of my DTH's that I modified from the SmartThings stock one to do just that:

https://github.com/vseven/SmartThings_VSeven/blob/master/devicetypes/vseven/enhanced-smartsense-temp-humidity-sensor.src/enhanced-smartsense-temp-humidity-sensor.groovy

See the "preferences" section and then lines 102 and 108.

Tx-ST-Home, 2018-06-06:
@vseven Thank you for you timely response to my questions. Guess its time for this noob to get some experience in programming with groovy and the use of Github. 

Just a quick follow-up would your  second handler "enhanced-smartsense-temp-humidity-sensor.groovy" be capable of adjusting the offsets in the SmartApp or do the offsets need to be hard coded into the handler. 
Also, while not related to offsets would you have any insights on why the Pin assignment for the DTH device isn't shown in the Konnected Sensor Status as shown in my attached files.

vseven, 2018-06-07:
The preference section I was referring to will let you set the offsets per device. As for the pin assignment I'm not real sure, I've been running Konnected for a while and I'm on a older version of the firmware and smart app without the temp device available.  I would make sure you have your smartapp updated.

closed 2022-08-23 by heythisisnate