konnected-io/konnected-security #62: Home assistant HTTP api password error

hugodelaroza, 2018-06-07:
When home assistant api has a password, konnected cannot inform sensors status to home assistant

heythisisnate, 2018-06-13:
I finally got an opportunity to explore this and found that the `api_password` is not an issue at all. The Konnected component API endpoint is configured to not require the password. Instead, an `access_token` is used from your Konnected config.

You may have been confusing this with a TLS issue that is being addressed here: #63

hugodelaroza, 2018-06-14:
or maybe I have a token, how can I know it?

heythisisnate, 2018-06-14:
You just make up any random string. It is synced to the Konnected device during discovery and used from that point to authenticate.

hugodelaroza, 2018-06-14:
but I allready have one:

konnected:
  access_token: skfn8sdfo8si54329
  devices:
    - id: 68c63a8a58ae
      binary_sensors:
        - zone: 1
          type: motion
          name: Pir 1 Konnected

This is my config.yaml

that else can I do?

rotcop4u2, 2018-06-14:
Is that with the indents? It should look like:

konnected:
  access_token: skfn8sdfo8si54329
  devices:
  - id: 68c63a8a58ae
  binary_sensors:
  - zone: 1
  type: motion
  name: Pir 1 Konnected

hugodelaroza, 2018-06-14:
I think it's not the indent, I get this in the serial connection:

Konnected firmware 2.2
NodeMCU 2.2.0.0 build 20180405 powered by Lua 5.1.4 on SDK 2.2.1(cfd48f3)
Heap:   39856   Initializing Konnected
Heap:   35312   Version:        2.2.0
Heap:   33688   Connecting to Wifi..
> Heap:         37536   Wifi connected with IP:         192.168.0.130   255.255.255.0   192.168.0.1
Heap:   37504   UPnP:   Listening for UPnP discovery
Heap:   37296   HTTP:   Starting server at http://192.168.0.130:17066
Heap:   37128   Loaded:         server
Heap:   32512   Initializing sensor pin:        1
Heap:   32200   Loaded:         application
E:M 1048
HTTP client: Disconnected with error: 46
HTTP client: Connection timeout
Heap:   34760   HTTP Call:      -1      state   1       pin     1

heythisisnate, 2018-06-14:
@hugodelaroza please try re-flashing the bin files from here: https://github.com/konnected-io/konnected-security/tree/homeassistant-fixes/firmware
If that doesn't fix the problem, post a new log

hugodelaroza, 2018-06-14:
can you tell me wich one should I flash first? or is it the same?

hugodelaroza, 2018-06-14:
NOW IT WORKS!!!!! Thank you very much!!!!

This weekend will migrate all the wires from the old alarm system to konnected!!

closed 2018-06-14 by heythisisnate