konnected-io/konnected-security #67: Relay activates/toggles when NodeMCU is rebooted, or reconnected to the internet

satyajitbagul, 2018-07-27:
Hi Nate,
I recently added a NodeMCU and 5V relay to my garage door along with an open/close sensor.
The setup worked very well.
The only problem that I am facing right now is whenever NodeMCU reboots, reconnects or we make changes to the pins in Konnected(connect) app the device restarts and connect again to the WiFi and it also **toggles/activates** the replay. This causes the garage door to open (when it should stay closed).
I was just wondering if there is fix that can be added the code which will recognize the last state of of actuator and not change it during reboot.

Hardware-
Node MCU - ESP8266 Development Board NodeMCU Lua V3 WIFI with CH340G USB and Base Shield
5V replay - 2 Channel Relay Module With Optocoupler 5V Active Low Level Amplifier Trigger

Connections-
**VCC** and **GND** of relay connected to NodeMCU **5v** and **GND** 
**IN2** pin of relay connected to NodeMCU **PIN8**
**"+"** of garage door connected to **NO2** of replay
**"-"** of garage door connected to **COM2** of replay

Setting in APP-
Low Level Trigger - Enabled
Momentary Delay - 500 milliseconds

heythisisnate, 2018-07-27:
You cannot use pin D8 with a low level relay, because D8 must be pulled LOW on boot which will cause the relay to trigger. The workaround is to use any other pin, or to use a high-level trigger relay on pin D8: https://konnected.io/collections/in-stock-now/products/1-channel-5v-relay-module-with-high-low-level-trigger

There's no software fix for this issue, as D8 is automatically pulled LOW on boot by the ESP.

closed 2018-07-27 by heythisisnate

satyajitbagul, 2018-07-27:
Thanks for the quick response. I will try to change it according to your recommendation.

satyajitbagul, 2018-07-27:
It is working perfectly now.
I **enabled** high level trigger and swapped the **"+"** of garage opener to **"NC2".**
Thanks for the help