konnected-io/konnected-security #71: Panic Error for Pin 8, both momentary switch and siren

satyajitbagul, 2018-09-30:
Hello Nate,

I recently updated 3 nodeMCUs with latest 2.2.4 release. 
the first nodemcu worked perfectly without any problems, i only have contact sensors setup on this.

second one has pin8 setup as siren and third has pin8 assigned as momentary switch. for both of these i am getting following error.

Konnected firmware 2.2.4
NodeMCU 2.2.0.0 build 20180917 powered by Lua 5.1.4 on SDK 2.2.1(6ab97e9)
Heap: 	39688	Initializing Konnected (84f3eb0dc90b)
Heap: 	33992	Version: 	2.2.4
Heap: 	32320	Connecting to Wifi..
> Heap: 	37136	Wifi connected with IP: 	10.0.0.20	255.255.255.0	10.0.0.1
Heap: 	35928	HTTP: 	Starting server at http://10.0.0.20:8903
Heap: 	35920	Loaded: 	server
Heap:	28728	Initializing sensor pin:	5
Heap:	28696	Initializing sensor pin:	1
Heap:	28696	Initializing sensor pin:	6
Heap:	28696	Initializing sensor pin:	9
Heap:	28696	Initializing sensor pin:	7
Heap:	28696	Initializing sensor pin:	2
Heap:	28336	Endpoint:	https://graph-na04-useast2.api.smartthings.com:443/api/smartapps/installations/****************************
Heap: 	28304	Loaded: 	application
Heap:	32920	HTTP Call:	200	state	userdata: 00000000	pin	8
PANIC: unprotected error in call to Lua API (/opt/nodemcu-firmware/local/fs/application.lua:122: bad argument #2 to 'write' (number expected, got userdata))

 ets Jan  8 2013,rst cause:2, boot mode:(3,6)

load 0x40100000, len 25784, room 16 
tail 8
chksum 0xd8
load 0x3ffe8000, len 2344, room 0 
tail 8
chksum 0x03
load 0x3ffe8928, len 136, room 0 
tail 8
chksum 0xca
csum 0xca
Œãäƒoì›;ƒûg|ìlld$`#“Ÿ|{Û$œnàƒnâ

heythisisnate, 2018-09-30:
@satyajitbagul this issues is fixed in the SmartThings Device Handlers. Please update your smartapps & device handers and I think that'll solve it.

satyajitbagul, 2018-10-08:
@heythisisnate, after updating all my device handlers it solved the problem.
thanks for the help.

closed 2018-10-08 by satyajitbagul