konnected-io/konnected-security #73: HA v0.81.2 update

01kbpatel, 2018-10-31:
After updating to the latest version of HA v0.81.2, the following error is reported

```
Traceback (most recent call last):
  File "/usr/local/lib/python3.6/site-packages/homeassistant/setup.py", line 145, in _async_setup_component
    hass, processed_config)
  File "/usr/local/lib/python3.6/site-packages/homeassistant/components/konnected.py", line 159, in async_setup
    ConfiguredDevice(hass, device).save_data()
  File "/usr/local/lib/python3.6/site-packages/homeassistant/components/konnected.py", line 241, in save_data
    DOMAIN, {'device_id': self.device_id})
TypeError: load_platform() missing 1 required positional argument: 'hass_config'
```
Any ideas on how to fix this?

nnhhuu, 2018-10-31:
i had to downgrade to 0.81.0 (it could be to 0.81.1, but to wanted to go for sure)  until the problem is solved... that's my recommendation.

heythisisnate, 2018-10-31:
This was broken due to a Home Assistant core change in 0.81.2 and quickly identified and fixed, see: https://github.com/home-assistant/home-assistant/issues/18003

The fix will be available in 0.81.3 soon. until then, stay on 0.81.1 or earlier.

closed 2018-10-31 by heythisisnate