konnected-io/konnected-security #79: Usage on ESP32?

AddoSolutions, 2019-01-04:
Hello, I am interested in using this on an ESP32 board for the added GPIO, performance, ethernet, etc.

Has anyone tried this? Do we know what it looks like to make this happen?  I am a NodeJS programmer and could help as needed (though I know lua is a lot different)

I saw some distant chatter about it but just wanted to check on it for real.

Thanks for the awesome software!

(and yes, I accidentally posted this before typing a message and somehow GitHub let me send it 🤦‍♂️ )

heythisisnate, 2019-01-17:
This is on our to-do list! It should be possible but will require some updates to the firmware and software to make use of the expanded capabilities of the ESP32.

AddoSolutions, 2019-06-29:
Do we have any idea of what it looks like, to just use it as-is with an ESP32? I would really like to be able to run these succors over ethernet, and even more so with POE.  

Something like this bad boy:
https://www.olimex.com/Products/IoT/ESP32/ESP32-POE/open-source-hardware

heythisisnate, 2019-06-30:
We are developing a product that will use the ESP32 with Ethernet and PoE capability. It's not ready yet but hopefully by the end of this year.

Currently the firmware does not support Ethernet, but that's in the works, too. See: https://github.com/nodemcu/nodemcu-firmware/issues/2796

rotcop4u2, 2019-06-30:
I was hesitant on purchasing due the possibly of wifi/bluetooth/zigbee/z-wave jammers.. you will definitely have a customer when this is released!!..

AddoSolutions, 2019-07-01:
@heythisisnate Great! I’m going to see if I can get the Ethernet part moving here. 

I should ask, the new product, do you guys still plan on making the codebase all open source like it is? Or were you going to close it off?

heythisisnate, 2019-07-01:
The device firmware, application code, and integrations will continue to be open source. Our custom hardware design will not be.

RichieRich69, 2019-07-23:
#Following Definately interested in an ethernet implementation of the project and esp32 offers so much more than esp8266.

Colmrenault, 2019-08-10:
Super, looking forward to ethernet capability also for my 24 zone setup.

zombodotcom, 2019-10-30:
Could we get the source code for the firmware?

heythisisnate, 2019-10-30:
@zombodotcom Due to the higher than expected cost of R&D for the ethernet-enabled ESP32 product, we've decided not to open-source the Konnected embedded application for the ESP32. The underlying firmware is still based on the nodemcu-firmware project and is open source: https://github.com/nodemcu/nodemcu-firmware/tree/dev-esp32

AddoSolutions, 2019-10-30:
Hey Nate, super disappointed it’s not open source, but which part is closed since I se the source in the link you sent?

heythisisnate, 2019-10-30:
The Konnected application for the ESP32 will not be open source.
We'll continue to maintain and improve the open-source Konnected application for the ESP8266 here: https://github.com/konnected-io/konnected-security

gomez85, 2020-06-25:
Hi @heythisisnate, I see that the Konnected Alarm Panel Pro will be on a ESP32 chip.. can we'll able to use the analog input to read the value of the resistor for discriminate what event came? For eg short circuit, masking ecc

btannous, 2020-07-02:
thats a neat idea.  For a zone with multiple switches, you could try to read the resistance to overload the switch and determine the exact switch

heythisisnate, 2020-07-03:
@gomez85 We've considered adding the capability to monitor an analog input with the ability to discern different resistance values. It will not be included in the initial release, but we may decide to add this feature in a future software update.

jswingen, 2021-06-11:
Have you considered offering the Konnected application for ESP32 as a software only or subscription since it is not going to be open source?

heythisisnate, 2021-06-11:
@jswingen Yes, the Konnected Pro software/firmware is freely available as a flashable binary here: https://help.konnected.io/support/solutions/articles/32000029848-alarm-panel-pro-release-notes

jswingen, 2021-06-11:
Oh great.  I misunderstood when you said it was not going to be open source.  Great news.  Thanks Nate.