konnected-io/konnected-security #80: Google Mesh Wifi

corystew, 2019-01-09:
The Luna MCU won't connect to a Google Mesh WIFI.

heythisisnate, 2019-01-17:
We have not been able to reproduce this issue. Google WiFi works great as far as we know. Several of us on the Konnected team (and MANY Konnected customers) are using Google WiFi with no issues at all.

I think your issue is unrelated to Google WiFi.

closed 2019-01-17 by heythisisnate

corystew, 2019-01-17:
I agree the issue may not be directly related to Google WiFi however if you have an existing IP network and add Google WiFi to it it creates another network which then causes the issue. The issue is that the node device has to be on the same IP network as the Samsung hub. I assume since the networking is basic in the node device it has this limitation. Is there a road map item to enhance the networking capability of the node device?

Regards,
Cory Stewart

Closed #80<https://github.com/konnected-io/konnected-security/issues/80>.

heythisisnate, 2019-01-17:
The issue you're describing is known a "Double-NAT". More info and suggested workarounds here:
https://support.google.com/wifi/answer/6277579?hl=en

Konnected & SmartThings uses the SSDP (Simple Service Discovery Protocol) to discover the devices on your local network. By design, SSDP traffic is limited only to a single subnet/network. It's not a limitation of the networking capabilities of the Konnected device, but rather the design of local LAN discovery that's causing the problem you're describing.

I recommend that you address the double-NAT situation and then you should be good to go.

corystew, 2019-01-18:
I have it resolved by plugging my Smartthings hub into one of the Google WiFi access points. Is the local LAN discovery going to be enhanced to function on more than the local network?

Regards,
Cory Stewart

https://support.google.com/wifi/answer/6277579?hl=en

heythisisnate, 2019-01-18:
No, it is not possible to "enhance". SSDP is a standardized protocol and by design it only works on the immediate subnet.

corystew, 2019-01-18:
OK. Thank you for the information and feedback on this. I don't prefer to have my Smartthings hub on the wireless network but it seems at this time it is the only choice.

Regards,
Cory Stewart

heythisisnate, 2019-01-18:
A better solution may be to bridge your multiple networks into one. Unless you have specific reasons for keeping two separate networks off of the same ISP connection, there's no value in doing it that way.

corystew, 2019-01-18:
Google WiFi is not able to run in bridge mode with multiple access points running mesh. Is what it is for now...

Regards,
Cory Stewart