konnected-io/konnected-security #82: After flashing Firmware 2.2.5 Esplorer still shows 2.2.4

qu3bec, 2019-01-17:
Not sure if this is an issue but just thought I'll put it here. Just flashed 2.2.5 and after flashing Esplorer still shows 2.2.4. Is that normal?
Konnected firmware 2.2.4
NodeMCU 2.2.0.0 build 20180917 powered by Lua 5.1.4 on SDK 2.2.1(6ab97e9)
Heap: 33568	Initializing Konnected (68c63ae19c2a)
Heap: 32376	Version: 2.2.5
Heap: 30728	Connecting to Wifi…
Heap: 30344	WiFi not configured
Heap: 30312	Entering Wifi setup mode
Heap: 23496	WiFi Setup started

heythisisnate, 2019-01-17:
You are right, I simply forgot to update the firmware version identifier.
updating now ...

heythisisnate, 2019-01-17:
Updated: https://github.com/konnected-io/konnected-security/releases/tag/v2.2.5

closed 2019-01-17 by heythisisnate