konnected-io/konnected-security #84: HTTP client: Disconnected with error: -8  HTTP client: Connection timeout

qu3bec, 2019-01-19:
Heap: 30480 HTTP Call: 204 state 0 pin 7
**HTTP client: Disconnected with error: -8
**HTTP client: Connection timeout****
Heap: 32240 HTTP Call: -1 state 1 pin 7
ets Jan  8 2013,rst cause:2, boot mode:(3,6)

I have 3 board, one connected to 5 motion sensors and a siren, other one connected to garage door opener,temp sensor & siren and a third test board connected only to 1 motion sensor. Board with motion sensors reboots very often, maybe every few hours and above is the log. 2nd board with 1 motion sensor reboots too but not that often,maybe because it only has 1 motion sensor. 3rd board garage door one, is rock solid. Please let me know if and how I can do more troubleshooting or provide more logs. It's a bit hard to capture on esplorer,had to leave a laptop connected.
Thanks

qu3bec, 2019-01-19:
[image]<br /><a target='_blank' href='https://movieplotholes.com/jurassic-park'>jurassic park fail</a><br />

Just rebooted again 10 mins ago after being up for nearly an hour. Sorry have attached as couldn't copy and paster the gibberish code shown in esplorer after rebooting, wasn't sure if it was important.

ceandraka, 2020-05-07:
I have similar. Nate indicated that motion sensor board needs to connect to HA more often (every time motion is sensed). If network issue at that time, it reboots to try to re-acquire network. Others never see the network issue because they are not communicating at the time. Could be router, WiFi, or other network issue. My Konnected cards are in a metal box, which is likely the issue. WiFi reports at about 50% strength at konnected boards.

ceandraka, 2020-05-22:
I made system mounting changes to improve WiFi signal strength to 70% or so, but still get reboots. As soon as it reboots, connection is good. Not sure where the problem is, nor how to further diagnose it. Was hoping the wifi strength was the issue.

closed 2022-06-17 by heythisisnate