konnected-io/konnected-security #86: E:M 32 PANIC: unprotected error in call to Lua API (error loading module 'httpd_req' from file 'httpd_req.lc': 	not enough memory)

qu3bec, 2019-02-17:
E:M 32
PANIC: unprotected error in call to Lua API (error loading module 'httpd_req' from file 'httpd_req.lc':
	not enough memory)
Reboots every time Siren goes off or if I try to test by testing siren. Tested on two boards both running 2.2.6
Above is what ESplorer shows when I test the siren.

heythisisnate, 2019-02-18:
Thanks for reporting this. Could you post the full boot output from your device?
Also, do you have the same problem if you downgrade to 2.2.5?

qu3bec, 2019-02-18:
Here are the logs and it's same after downgrading. Though it seems to give different error messages on different pins. Logs are when connected to D8 Pin. First set from 2.2.6 and the 2nd one from 2.2.5
Communication with MCU..Got answer! Communication with MCU established.
AutoDetect firmware...

Can't autodetect firmware, because proper answer not received (may be unknown firmware). 
Please, reset module or continue.
ŒbÄûg'Ÿlggœãìcp„‡l{drlpògàƒlcoâ|„„‡Ž�cŒó'oç
Konnected firmware 2.2.6
NodeMCU 2.2.0.0 build 20190203 powered by Lua 5.1.4 on SDK 2.2.1(6ab97e9)
Heap: 	34768	Initializing Konnected (68c63ae19c2a)
Heap: 	33152	Version: 	2.2.6
Heap: 	31496	Connecting to Wifi..
> Heap: 	36328	Wifi connected with IP: 	192.168.1.5	255.255.255.0	192.168.1.1
Heap: 	35128	UPnP:	Listening for UPnP discovery
Heap: 	34760	HTTP: 	Starting server at http://192.168.1.5:22785
Heap: 	34600	Loaded: 	server
Heap:	24448	Initializing sensor pin:	6
Heap:	24416	Initializing actuator pin:	5
Heap:	24416	Initializing actuator pin:	8
Heap:	24008	Endpoint:	https://graph-na04-useast2.api.smartthings.com:443/api/smartapps/installations/adca90da-380f-4dcd-a6be-07d3a3805088
Heap: 	23976	Loaded: 	application
Heap:	29240	HTTP Call:	200
Heap:	29200	Initialized actuator Pin:	5	Trigger:	1	Initial state:	0
Heap:	29024	HTTP Call:	200	state	0	pin	8
Heap:	28960	Initialized actuator Pin:	8	Trigger:	1	Initial state:	0
Heap:	28744	HTTP Call:	204	state	1	pin	6
Heap:	28648	HTTP Call:	204	state	0	pin	6
E:M 320
PANIC: unprotected error in call to Lua API (not enough memory)

 ets Jan  8 2013,rst cause:2, boot mode:(3,6)

load 0x40100000, len 26044, room 16 
tail 12
chksum 0xc3
ho 0 tail 12 room 4
load 0x3ffe8000, len 2340, room 12 
tail 8
chksum 0x61
load 0x3ffe8924, len 136, room 0 
tail 8
chksum 0x3b
csum 0x3b
„ãäÃgì“{ƒûg|ädlll`c›Ÿ|;’dœoà‚gã
Konnected firmware 2.2.6
NodeMCU 2.2.0.0 build 20190203 powered by Lua 5.1.4 on SDK 2.2.1(6ab97e9)
Heap: 	34768	Initializing Konnected (68c63ae19c2a)
Heap: 	33152	Version: 	2.2.6
Heap: 	31496	Connecting to Wifi..
> Heap: 	36328	Wifi connected with IP: 	192.168.1.5	255.255.255.0	192.168.1.1
Heap: 	35128	UPnP:	Listening for UPnP discovery
Heap: 	34760	HTTP: 	Starting server at http://192.168.1.5:22785
Heap: 	34600	Loaded: 	server
Heap:	24448	Initializing sensor pin:	6
Heap:	24416	Initializing actuator pin:	5
Heap:	24416	Initializing actuator pin:	8
Heap:	24008	Endpoint:	https://graph-na04-useast2.api.smartthings.com:443/api/smartapps/installations/adca90da-380f-4dcd-a6be-07d3a3805088
Heap: 	23976	Loaded: 	application
Heap:	29056	HTTP Call:	200
Heap:	29024	Initialized actuator Pin:	5	Trigger:	1	Initial state:	0
Heap:	28520	HTTP Call:	200	state	0	pin	8
Heap:	28448	Initialized actuator Pin:	8	Trigger:	1	Initial state:	0
Heap:	28528	HTTP Call:	204	state	1	pin	6
Heap:	28528	HTTP Call:	204	state	0	pin	6
E:M 320
PANIC: unprotected error in call to Lua API (not enough memory)

 ets Jan  8 2013,rst cause:2, boot mode:(3,6)

load 0x40100000, len 26044, room 16 
[... 206 lines cut ...]
Heap:	31832	HTTP Call:	204	state	0	pin	6
Heap: 	27328	Responding to UPnP Discovery request from 192.168.1.7:48400
Heap:	31928	HTTP Call:	204	state	1	pin	6
Heap:	31824	HTTP Call:	204	state	0	pin	6
Heap:	31840	HTTP Call:	204	state	1	pin	6
Heap:	31824	HTTP Call:	204	state	0	pin	6
Heap:	31904	HTTP Call:	204	state	1	pin	6
E:M 24
PANIC: unprotected error in call to Lua API (error loading module 'httpd_res' from file 'httpd_res.lc':
	not enough memory)

 ets Jan  8 2013,rst cause:2, boot mode:(3,6)

load 0x40100000, len 25736, room 16 
tail 8
chksum 0xb8
load 0x3ffe8000, len 2340, room 0 
tail 4
chksum 0xf4
load 0x3ffe8924, len 136, room 4 
tail 4
chksum 0x3f
csum 0x3f

This is on a test board. Basically relay won't get activated when motion is being detected. Board errors out and app shows activating but then goes back to Alarm Off