konnected-io/konnected-security #90: Homeassistant .90- Error

THland83, 2019-03-20:
X-Post from support forum. 

Error doing job: Future exception was never retrieved

Traceback (most recent call last):
  File "/usr/local/lib/python3.7/concurrent/futures/thread.py", line 57, in run
    result = self.fn(*self.args, **self.kwargs)
  File "/usr/local/lib/python3.7/site-packages/homeassistant/components/konnected/__init__.py", line 131, in device_discovered
    setup_device(host, port)
  File "/usr/local/lib/python3.7/site-packages/homeassistant/components/konnected/__init__.py", line 121, in setup_device
    discovered.setup()
  File "/usr/local/lib/python3.7/site-packages/homeassistant/components/konnected/__init__.py", line 284, in setup
    self.sync_device_config()
  File "/usr/local/lib/python3.7/site-packages/homeassistant/components/konnected/__init__.py", line 388, in sync_device_config
    if self.desired_settings_payload() != self.current_settings_payload():
  File "/usr/local/lib/python3.7/site-packages/homeassistant/components/konnected/__init__.py", line 368, in current_settings_payload
    settings = self.status['settings']
KeyError: 'settings'

hunterjm, 2019-03-22:
I am getting this error as well.  I am a couple firmware versions behind.

TSparkas, 2019-03-22:
+1. I have the same error on HA 0.90.1. Latest Konnected firmware on NodeMCU. So it's related to recent HA changes, component needs to be updated.

01kbpatel, 2019-03-22:
+1, works perfectly v0.89.2, but as soon as you upgrade get errors in the log, had to go back to old version until component is fixed in new update

heythisisnate, 2019-03-23:
This appears to be an issue with the HA component or changes in the latest version of HA. I am looking into it. Since it's not a firmware issue, I'm closing this issue in favor of this one on the home-assistant project: https://github.com/home-assistant/home-assistant/issues/22257

closed 2019-03-23 by heythisisnate