konnected-io/konnected-security #94: lua: cannot open wifi.lua and No LFS image loaded

amorozzo19, 2019-05-08:
Hello Nate! Thanks for putting this project together. I am hopping you can take a look at some errors I am getting when trying to upload the files in the src folder:

Konnected firmware 2.2.7
NodeMCU 2.2.0.0 build 20190424 powered by Lua 5.1.4 on SDK 2.2.1(6ab97e9)
> Uploading to ESP file actuators.lua...Success
> Uploading to ESP file dht_sensors.lua...Success
> Uploading to ESP file ds18b20_sensors.lua...Success
> Uploading to ESP file enduser_setup.html.gz...Success
> Uploading to ESP file http_favicon.ico.gz...Success
> Uploading to ESP file http_index.html.gz...Success
> Uploading to ESP file init.lua...Success
> Uploading to ESP file sensors.lua...Success
> Uploading to ESP file settings.lua...Success

> rl $
No LFS image loaded

Konnected firmware 2.2.7
NodeMCU 2.2.0.0 build 20190424 powered by Lua 5.1.4 on SDK 2.2.1(6ab97e9)
Heap: 	40048	Initializing Konnected (807d3a3e0d52)
Heap: 	37008	Compiling: 	ssdp.lua
Heap: 	37008	Compiling: 	ssdp_notify.lua
Heap: 	33200	Compiling: 	variables_build.lua
Heap: 	33400	Compiling: 	start.lua
Heap: 	33896	Compiling: 	dht_sensors.lua
Heap: 	33504	Compiling: 	application.lua
Heap: 	37008	Compiling: 	wifi.lua
Heap: 	30832	Compiling: 	device.lua
Heap: 	35152	Compiling: 	server_device.lua
Heap: 	28832	Compiling: 	variables_set.lua
Heap: 	34480	Compiling: 	server.lua
Heap: 	37008	Compiling: 	httpd_req.lua
Heap: 	32880	Compiling: 	httpd_res.lua
Heap: 	30824	Compiling: 	wipe.lua
Heap: 	33144	Compiling: 	server_status.lua
Heap: 	33416	Compiling: 	sensors.lua
Heap: 	33656	Compiling: 	led_flip.lua
Heap: 	32704	Compiling: 	server_settings.lua
Heap: 	32080	Compiling: 	server_receiver.lua
Heap: 	32152	Compiling: 	ds18b20_sensors.lua
Heap: 	36592	Compiling: 	actuators.lua
Heap: 	36200	Compiling: 	settings.lua
Heap: 	35896	Compiling: 	ssdp_response.lua
Heap: 	36384	Application Version: 	2.2.7
lua: cannot open wifi.lua
stack traceback:
	[C]: in function 'dofile'
	init.lua:16: in main chunk
	[C]: ?
> 

Then I noticed that you moved most or all the lua files to the lfs folder, so I uploaded those files instead, and got this error:

> Uploading to ESP file application.lua...Success
> Uploading to ESP file device.lua...Success
> Uploading to ESP file httpd_req.lua...Success
> Uploading to ESP file httpd_res.lua...Success
> Uploading to ESP file init.lua...Success
> Uploading to ESP file led_flip.lua...Success
> Uploading to ESP file server.lua...Success
> Uploading to ESP file server_device.lua...Success
> Uploading to ESP file server_receiver.lua...Success
> Uploading to ESP file server_settings.lua...Success
> Uploading to ESP file server_status.lua...Success
> Uploading to ESP file ssdp.lua...Success
> Uploading to ESP file ssdp_notify.lua...Success
> Uploading to ESP file ssdp_response.lua...Success
> Uploading to ESP file start.lua...Success
> Uploading to ESP file variables_build.lua...Success
> Uploading to ESP file variables_set.lua...Success
> Uploading to ESP file wifi.lua...Success
> Uploading to ESP file wipe.lua...Success

> s$
No LFS image loaded

Konnected firmware 2.2.7
NodeMCU 2.2.0.0 build 20190424 powered by Lua 5.1.4 on SDK 2.2.1(6ab97e9)
> 

** Note ** I changed the file name from _init.lua to just init.lua.

The LFS image is not in the scr/lfs location as indicated in the "spiffs.lst" under scripts

Thanks in advance for your help!

closed 2019-05-09 by amorozzo19

viperk1, 2019-06-07:
Did you solve this? I'm running into the same thing so if you figured it out I would love to know what you did.

heythisisnate, 2019-06-07:
@viperk1 you should be flashing the pre-built images that we provide in the /firmware directory

Unless you are customizing code, there's no need to upload files individually anymore.

viperk1, 2019-06-10:
@heythisisnate Thanks for the assist! I've got it running after flashing the filesystem (that's what I was missing before). Worked quite flawlessly after that although I still have to power and connect the motion sensors and siren/relay. Will definitely be donating to the project now that I've seen it working for me!