konnected-io/konnected-security #98: Konnected Interface: SmartThings latency

daltskin, 2019-08-01:
Can anything be done/tweaked to help reduce the latency from a motion sensor being triggered to turning on lights.

Using a Konnected Interface on a Honeywell Galaxy 2-20
SmartThings Routine to turn on Hue lights

The current delay is around 3-5 seconds before lights come on.

heythisisnate, 2019-08-01:
At this time, the only option is to switch to Home Assistant or Hubitat, both of which run locally and respond much faster.
We're working on a new SmartThings integration that should be a bit faster. It will be ready later this year.

closed 2019-08-01 by heythisisnate

daltskin, 2019-10-27:
Hey @heythisisnate any updates on the new ST integration?

heythisisnate, 2019-10-27:
It is almost ready! I have it working very well in our test environment. And yes, the response times are _much_ faster. I'm working with SmartThings next week to start the validation and approval process to get it built into the SmartThings app.

daltskin, 2019-10-28:
Sounds good! Really looking forward to trying this out.  Any firmware updates requires on the konnected hardware, or just updates within the ST app?

If you need any beta testers lmk :)

heythisisnate, 2019-10-28:
Yes, it will require a firmware update as well as a completely new ST integration using the new ST app. We can't really do a public beta because SmartThing's platform doesn't allow for that. We're planning on beginning the ST certification/approval process in the coming weeks.