konnected-io/konnected-security #99: MQTT support?

podarok, 2019-08-03:
Hi.
Thank you for the device. Received recently and replaced Integra64 .
I see that devices are sending information to Hubitat every time there is a change in sensors.
Is there a plan to add MQTT support?
Would be cool to use this protocol. It would be easier to add monitoring later

heythisisnate, 2019-08-04:
Yes, MQTT support is on the near-term roadmap.

Korte68, 2019-12-15:
great...any estimation?

lipigab, 2020-02-03:
That's the only thing while I haven't pulled the trigger yet...

nzbullet, 2020-04-16:
Any update on when this will be available? I’m a Homebridge user and don’t user any of the current integration options (and not do I intend to) so like others not having MQTT it is the only thing that stops me buying.

Korte68, 2020-04-17:
> Yes, MQTT support is on the near-term roadmap.

we have different meaning on 'near-term roadmap'....

mkormendy, 2020-07-17:
@heythisisnate,... any updates?

nzbullet, 2020-08-26:
Any updates on this 'near-term' roadmap item?

heythisisnate, 2020-08-26:
Ah yeah, this has definitely been pushed out longer than I originally expected. Priorities shift.

I'm still considering adding MQTT support to the Konnected firmware, but also recommending ESPHome as an alternative firmware that already supports it well. All Konnected hardware will run ESPHome. You can find a zone to NodeMCU pin mapping here: https://help.konnected.io/support/solutions/articles/32000026808-pin-mapping

Then, use the NodeMCU platform in your ESPHome configuration. An example config may look like this:

```
esphome:
  name: konnected-1
  platform: ESP8266
  board: nodemcuv2

mqtt:
  broker: 10.0.0.2
  username: heythisisnate
  password: MyMQTTPassword

binary_sensor:
  - platform: gpio
    name: "Front Door"
    pin: D1
  - platform: gpio
    name: "Patio Door"
    pin: 
      number: D2
      mode: INPUT_PULLUP
  - platform: gpio
    name: "Living Room Motion"
    pin:
      number: D5
      mode: INPUT_PULLUP
    device_class: motion

switch:
  - platform: gpio
    pin: D8
    name: "Siren"
```

Please let me know if the ESPHome approach would meet your needs. I'm interested in your use cases for MQTT support. It's not a popular feature request, but a few really want it.

emlowe, 2020-10-16:
MQTT support would allow connected to Homeseer automation without the need for a dedicated homeseer plugin (which doesn't exist) - as one could use the existing MQTT homeseer plugin. This would also apply to any other automation hub not currently supported, as MQTT support is pretty ubiquitous on such hubs. One could also write/use custom Node-Red implementations. It's just very flexible. ESPHome is an alternative that I'll look at (although it seems odd to purchase konnected.io and then overwrite the firmware with a 3rd party one)

heythisisnate, 2020-10-16:
@emlowe Fully agree with the benefits of MQTT for platforms like Homeseer. You're absolutely right on all counts.
After scoping it out what it would take to add MQTT to our existing firmware, we came to the conclusion that ESPHome's current capabilities would exceed anything that we could build in a minimal implementation, and therefore is a better option than rolling our own MQTT support. Since our devices use the official ESP-12F or ESP-32 chips and already have USB-serial access, ESPHome will work perfectly with our hardware. You're still getting the benefits of Konnected hardware, including built-in lightning protection, short-circuit protection, overcurrent protection, high build quality assembled in the USA, 1-year warranty and technical support even if you flash a 3rd-party firmware.

If you would like specific guidance on setting up MQTT with Konnected and ESPHome, please reach out to us at help@konnected.io. We have plans to publish a guide on our support site to do so, but this hasn't been done yet.

heythisisnate, 2021-03-07:
Finally published some documentation on this topic. Using ESPHome you can use all of the hardware features of Konnected via MQTT. More info: https://help.konnected.io/support/solutions/folders/32000036472

closed 2021-03-07 by heythisisnate

alexbk66, 2024-01-24:
https://forums.homeseer.com/forum/hs4-products/hs4-plugins/security-plug-ins-aa/ak-konnected-alarm-alexbk66/1649946-ak-konnected-alarm-panel-and-garage-door-opener-gdo