konnected-io/konnected-security #105: v2.3.0 boot freeze on NTP sync failure

LouPaloma, 2019-09-15:
Hooking up to a serial port monitor I see the boot messages: 
```
Konnected firmware 2.3.0 NodeMCU 2.2.0.0 build 20190805 powered by Lua 5.1.4 on SDK 2.2.1(6ab97e9) 
Heap: 41744 Initializing Konnected (dc4f2218ef45) 
Heap: 40680 Application Version: 2.3.0 
Heap: 39528 Connecting to Wifi.. > 
Heap: 42248 Wifi connected with IP: 172.16.x.x Gateway: 172.16.x.x
Heap: 40984 Time sync failed! 
```
There are no further messages and the device does not respond to the status endpoint. 

As you can see the WIFI connection is set and the information is verified correct via nmap. 

NOTE: MY SECURITY SYSTEM IS ON A PRIVATE NETWORK AND DOES NOT, _BY_DESIGN_, HAVE INTERNET ACCESS. I do have an NTP server on the network as well, but the device doesn't have that information configured. Time sync failure should not prohibit firmware boot, yes? 

Providing Internet access is not an option.

heythisisnate, 2019-09-16:
I confirmed that this is an issue. Thanks for reporting this @LouPaloma 

In 2.3.0 I added a step to the boot sequence to sync the current time via NTP.
It will first ask the gateway's IP (typically your router) for the current time, and then fall back to time.google.com and pool.ntp.org. If all three of these fail, the device fails to boot.

This is actually a fairly critical bug because if there's a internet outage and the device reboots, it may get stuck in this failed state and never recover.

heythisisnate, 2019-09-20:
This is fixed in release 2.3.1

closed 2019-09-20 by heythisisnate

LouPaloma, 2019-09-21:
Confirmed fix is working.

K-MTG, 2019-09-21:
Is it possible to preform an OTA update? Or will it be available soon?

eduardojacr, 2019-09-21:
It's possible to do it OTA.  I already did it. Make sure the Konnected app
is updated.  They updated today.
Regards
Jorge

heythisisnate, 2019-09-22:
Yes, if you've already manually flashed 2.3.0, you can OTA update to the 2.3.1 update containing this fix via the new Konnected mobile app.

It's available for Android here: https://play.google.com/store/apps/details?id=io.konnected.konnected

An iOS version should be hitting the App Store next week (pending approval by Apple).

KScottTerry, 2019-09-29:
Hi Nate, the OTA update was flawless and easy, (I like easy.) I watched the Nodes blue light blink a few times and after the suggested minute I refreshed. I checked and noticed the software updated to 2.3.1 but no numbers changed in the firmware; 2.3.0 . Please pardon my confusion, but I thought I read [somewhere] this was a firmware update, am I right? Just confirming the update was a success.
[image]

heythisisnate, 2019-09-29:
Yes, this is correct. Only the Konnected software (application code) can be
updated OTA. The "firmware" refers to the underlying os built on
NodeMCU-firmware and it hasn't changed.

dsaul560, 2019-10-04:
I can't find the iOS version of the app. Is that out yet?
When is the MQTT update coming? Can't wait! lol
Thanks!

lastoption, 2019-10-13:
Hi Nate, just downloaded the iOS app. When doing device discovery it only found two of four. Any ideas? The two it found updated to 2.3.1 flawlessly. Thanks!

Bwiestenberg, 2019-10-25:
After flashing to the latest firmware and file system,  the nodemcu chip is unresponsive.  Instead of a flashing blue LED and the wifi signal showing up when searched for,  I now get a blue led that flashes once then is solid on the whole time.  I've downloaded the firmware and filesystem over to eliminate errors,  and have reflashed the chips multiple times,  with the same results.  I selected erase all data when re flashing as well.

heythisisnate, 2019-10-27:
This is not a support forum. Issues unrelated to the topic of this bug will not be addressed here.