konnected-io/konnected-security #111: Support non-discovery based homeassistant setup

greghaynes, 2019-10-19:
AFAICT, the only method supported for connecting to homeassistant is via upnp and homeassistant's discovery system. This restricts supported networking configurations between konnected and homeassistant to those where they both reside on the same layer 2 network and can share multicast data. One example of where this does not work is a configuration where konnected and homeassistant are on separate networks (this is the case if homeassistant is running in docker).

It would be great to support a non-discovery based setup method. e.g. allowing a user to manually enter either the homeassistant api info as part of configuring konnected, or allowing a user to configure homeassistant to know the uri to reach konnected.

heythisisnate, 2019-10-19:
It's already supported.
Just set _host_ and _port_ options in the device config. See docs: https://www.home-assistant.io/integrations/konnected

Each device uses a different HTTP port for the embedded server. You can find this out by using our mobile app to discover the port, or connect the device to you computer by USB.

closed 2019-10-19 by heythisisnate

greghaynes, 2019-10-19:
Awesome, that worked great. Thanks!