konnected-io/konnected-security #114: Board trying outbound connection to 23.129.64.159!

qu3bec, 2019-12-14:
Just noticed these entries on the Firewall running on my router - 
Dec 11 21:01:34 kernel: [BLOCKED - OUTBOUND] IN=br0 OUT= MAC=18:31:bf:4a:20:e0:68:c6:3a:e1:9c:2a:08:00 SRC=192.168.1.11 DST=23.129.64.159 LEN=76 TOS=0x00 PREC=0x00 TTL=255 ID=12 PROTO=UDP SPT=4096 DPT=123 LEN=56

Wasn't sure if this is the right place for this issue or is it even an issue as these connections are getting blocked,

qu3bec, 2019-12-14:
Any reason why konnected board would try utbound connection to 23.129.64.159? Noticed this in the logs on the firewall running on my router. 
Dec 11 21:01:30 kernel: [BLOCKED - OUTBOUND] IN=br0 OUT= MAC=18:31:bf:4a:20:e0:68:c6:3a:e1:9c:2a:08:00 SRC=192.168.1.11 DST=23.129.64.159 LEN=76 TOS=0x00 PREC=0x00 TTL=255 ID=6 PROTO=UDP SPT=4096 DPT=123 LEN=56

rotcop4u2, 2019-12-14:
By using google on the IP address. It shows that is a tor IP. Also shows
that it is an abusive IP.

heythisisnate, 2019-12-14:
No idea why Konnected would be hitting this ip. Are you sure the source is a konnected device?

qu3bec, 2019-12-15:
> No idea why Konnected would be hitting this ip. Are you sure the source is a konnected device?

Yes I've set up the board with a static IP 192.168.1.11

DeeCla1973, 2019-12-15:
The alarm can't seem to connect to my wifi as of yesterday.  I have no idea
and can't see to correct the issue.

heythisisnate, 2019-12-15:
I really have no idea. This is not much info to go on. Maybe try capturing the traffic with wireshark to see if we can see what kind of content it is?

qu3bec, 2019-12-16:
> I really have no idea. This is not much info to go on. Maybe try capturing the traffic with wireshark to see if we can see what kind of content it is?
https://otx.alienvault.com/indicator/ip/23.129.64.159
Under Passive DNS some ntp servers listed. Does the board connect to ntp servers? I just found out that my TP-Link HS110 plug also tried to hit this ip!!

qu3bec, 2019-12-16:
https://otx.alienvault.com/indicator/ip/23.129.64.159
Under Passive DNS some ntp servers listed. Does the board connect to ntp servers? I just found out that my TP-Link HS110 plug also tried to hit this ip!!

heythisisnate, 2019-12-16:
Yes, it does connect to NTP to sync the current time.

heythisisnate, 2019-12-16:
Looks like `pool.ntp.org` is one of the DNS entries, so that makes sense now.

closed 2019-12-16 by heythisisnate

qu3bec, 2019-12-16:
Yeah it does. How often does the board sync current time? Mine reboots once every few days and sometimes twice a day so maybe it can't connect to ntp servers when it reboots?

heythisisnate, 2019-12-16:
It syncs after every reboot.
It will prefer to get the current time from your router/gateway. So, if you can configure your router to act as a NTP server, then you can avoid going out to the internet to sync time.