konnected-io/konnected-security #117: Relay triggers every time board reboots.

qu3bec, 2019-12-21:
After upgrading to 2.3 both garage door & siren turn on if board reboots. Only momentary switch For garage door and siren connected to board. It was all fine before upgrade. Low level trigger is checked in app for both devices. Has something changed in the firmware or file system?

qu3bec, 2019-12-21:
Did some more troubleshooting, why are the pins getting initialized on boot? This is what's causing the garage door to open & Siren to sound.
Heap:	38272	HTTP Call:	200	state	1	pin	6
Heap:	38208	Initialized actuator Pin:	6	Trigger:	0	Initial state:	1
Heap:	37768	HTTP Call:	200	state	1	pin	5
Heap:	37704	Initialized actuator Pin:	5	Trigger:	0	Initial state:	1

heythisisnate, 2020-02-27:
What type of relay are you using? Make sure the the relay trigger is also set to L(ow)

txgi108, 2020-02-29:
> What type of relay are you using? Make sure the the relay trigger is also set to L(ow)

I had the same issue, flashed the board to 2.3.3 and both relays activated after the reboot. Using a DIY Kit and a 5V-2 Channel relay with the trigger set to Low. Just to test it out, I cycled power to the board and the same thing happened on restart.

heythisisnate, 2020-02-29:
Make sure you have [re-synced device settings](https://help.konnected.io/support/solutions/articles/32000023498-re-sync-device-settings) after flashing

txgi108, 2020-02-29:
Yep, did that. The relays triggered after that...

 Fortunately the Konnected Smart Home app still has 2.3.1. After I reverted, the issue went away.

I have the relays running my garage doors, so I can't have them opening every time the board reboots...

heythisisnate, 2020-02-29:
hmm, will try to reproduce this

TazUk, 2020-03-06:
I've also come across this today - the relay triggers every time I restart HomeAssistant or reboot the Konnected board.
I'm using the relay as a momentary switch to arm and disarm my alarm, meaning that this is a major issue in my case.

Unfortunately I'm completely unable to downgrade my firmware from 2.3.3 - I've tried 7 different known good USB cables, and 5 different sets of drivers.
Is there an alternative method available to force a downgrade?

TazUk, 2020-03-07:
After trying 2 Windows 10 laptops, 7 usb cables and 5 different driver sets I managed to revert back to 2.3.1 using 
`curl -X POST -H "Content-Type: application/json"  -d '{"uri":"https://konnected-io.s3.us-east-2.amazonaws.com/esp8266-lfs/lfs-2.3.1.img"}'  http://<konnectedip>:<port>/ota`

Rolling back the firmware has completely resolved the issue.

heythisisnate, 2020-03-19:
I was able to reproduce and confirm that this is a bug.
It's fixed here: https://github.com/konnected-io/konnected-security/commit/d52b64983ae9bfbfe859766095cb314be00fe40e

This will be released in 2.3.4 within a few days.

The bug affects only low-level trigger relays, and causes them to momentarily trigger when the device is restarted.

closed 2020-03-19 by heythisisnate

TazUk, 2020-04-19:
Hi, 

My relay continues to trigger when rebooting the Konnected v2 add-on board and when restarting HomeAssistant.
Happening both with release 2.3.4 and release 2.3.5, and HA version 107.x and (current) version 108.6

heythisisnate, 2020-04-19:
@TazUk Please confirm the trigger setting in the Konnected configuration matches the trigger on your relay. The easiest way to view the Konnected device's configuration is to download the Konnected app from the Apple / Google Play store, scan your network for the device, and then tap on "View device status". The trigger setting is shown in the _actuators_ section of the status page.

0 = low
1 (or empty) = high

Make sure that your relay is configured accordingly. The relay modules that we sell have jumper to configure, but some relays may be fixed low or high.

Also -- what pin/zone do you have your relay connected to? Not all pins are compatible with high and low triggers.

TazUk, 2020-04-19:
@heythisisnate 

Nate, you were spot on.

Relay was jumpered to High, but my configuration was set to low.

After some messing around, I've reconfigured to 'High' in HA and checked that it passed through to the board from the app and ran a few restarts and all now behaves (relay triggers when board is first powered but I would expect that).

Interestingly trying to jumper the relay as low prevented the board from being accessible at all (from app or HA), so clearly high is the way to go for my setup. 

I'm using the "OUT" terminal to fire the relay and using zones 1 to 3 to read some programmable outputs from the Texecom alarm panel it's wired to.

-------------
Apologies for my mistake and thanks again for the assist. This has been driving me nuts for weeks so with hindsight was bound to be something stupid at my end.