konnected-io/konnected-security #121: hassio Konnected port

weswark, 2020-02-26:
With the recent home assistant .106 update I get a notification for discovering a new device which point to the konnected integration (this is already set up in my yaml and working. when clicking on the check it out link, it takes me to the below interface of which it asks for a konnected port........... how do I find out the konnected port?

"Please enter the host information for your Konnected Panel."
Konnected Device IP Address = No problem here
Konnected Device Port = How do I find this out? It is not documented on either the Konnected Website nor the "https://www.home-assistant.io/integrations/konnected/"

Thanks in advance

heythisisnate, 2020-02-27:
This should discover automatically on your network.
@kit-klein do you know why SSDP may not be working in this case?

kit-klein, 2020-02-27:
A couple things could cause this.  

First confirm that the [SSDP integration](https://www.home-assistant.io/integrations/ssdp/) is not disabled in your Home Assistant instance.  It's enabled by default so this should only be an issue if you manually disabled it.

Beyond that their are a variety of things that can impact ssdp.  Common sources of trouble are
missing mDNS repeaters on VLANs, Wi-Fi isolations, broadcast blocking, just a router bug.

If you think everything should be working you can test from a computer using the netdisco module
```
pip3 install netdisco
python3 -m netdisco
```

It will scan for a bit and then print out the results.  Among them should be an entry for your Konnected Alarm Panel
```
konnected:
[{'host': '192.168.1.78',
  'manufacturer': 'konnected.io',
  'model_name': 'Konnected Pro',
  'model_number': '0.3.1',
  'name': 'Konnected Pro',
  'port': 9123,
  'serial': '0x46840d8edf2b',
  'ssdp_description': 'http://192.168.1.78:9123/Device.xml',
  'udn': 'uuid:8f655392-a778-4fee-97b9-48259180x46840d8edf2b',
  'upnp_device_type': 'urn:schemas-konnected-io:device:Security:1'}]
```

weswark, 2020-02-27:
Hi All
Power cycling the konnected and then the hass seemed to force the auto config to work from my previously working yaml.
I was able to use the konnected ios app to check the port and make sure its talking to hass - this is awesome.
Working again, thanks all

closed 2020-02-27 by weswark