konnected-io/konnected-security #124: Constant blue light on board after upgrading to 2.3.3

qu3bec, 2020-03-24:
I'm getting constant blue light on board after upgrading to 2.3.3. Tried flashing many times again with Konnected Flash tool but no luck. Esplorer shows this - 
pc1=0x40212d34, epc2=0x00000000, epc3=0x00000000, excvaddr=0x00000000, depc=0x00000000
Fatal exception (0): 
epc1=0x40212d34, epc2=0x00000000, epc3=0x00000000, excvaddr=0x00000000, depc=0x00000000
Fatal exception (0): 
epc1=0x40212d34, epc2=0x00000000, epc3=0x00000000, excvaddr=0x00000000, depc=0x00000000
Fatal exception (0): 
epc1=0x40212d34, epc2=0x00000000, epc3=0x00000000, excvaddr=0x00000000, depc=0x00000000
Fatal exception (0): 
epc1=0x40212d34, epc2=0x00000000, epc3=0x00000000, excvaddr=0x00000000, depc=0x00000000
Fatal exception (0): 
epc1=0x40212d34, epc2=0x00000000, epc3=0x00000000, excvaddr=0x00000000, depc=0x00000000
Fatal exception (0): 
epc1=0x40212d34, epc2=0x00000000, epc3=0x00000000, excvaddr=0x00000000, depc=0x00000000
Fatal exception (0): 
Has the board gone faulty or is there a way to revive it?

qu3bec, 2020-03-24:
Issue can be closed as it might be an issue with my USB port or laptop but I tried two different laptops . @heythisisnate Please note I tried flashing with Konnected Flash tool many times and it didn't work, then tried with esptool.py using standard command mentioned in scripts/flash, also did not work. WHy would the Konnected Flash tool suddenly not work, I tried on two different laptops. 
This is the command that worked and it only works when flashing filesystem first then firmware. 
esptool.py --port=COM6 --baud 460800 write_flash --flash_mode dio --flash_size=detect 0 konnected-filesystem-0x100000-2-3-1.img

heythisisnate, 2020-03-24:
This problem can be caused by a filesystem flashed at the wrong memory location.
The _filesystem_ file needs to be flashed at address 0x100000. Command should look like this:
```
esptool.py --port=COM6  write_flash --flash_mode dio 0x100000 konnected-filesystem-0x100000-2-3-3.img
```

Next release we'll be combining the FW and FS images into one to make it simpler to flash all at once.

closed 2020-03-24 by heythisisnate