konnected-io/konnected-security #127: NodeMCU - No wifi after flashing

mshuflin, 2020-04-25:
When I flash konnected-firmware-2-3-5.bin and konnected-filesystem-0x100000-2-3-5.img using the Konnected Flashing Tool 3.1, I can't see any wifi broadcast from the device. It seems to not be on. I tried using tasmotizer to flash tasmota and it works and shows the wifi. Any idea what might be causing the NodeMCU wifi to not work with Konnected, but fine with other firmware?

bubez81, 2020-05-29:
same problem here!

heythisisnate, 2020-05-29:
If you had previously configured the wifi, the credentials are still retained and it's re-connecting automatically to your home network and will not start broadcasting.

Chose the "erase first" option to also erase the stored wifi credentials.

closed 2020-05-29 by heythisisnate

bubez81, 2020-05-31:
sadly it's a new installation...the problem is the os...I'm using macOS Catalina...with a windows pc it works perfectly