konnected-io/konnected-security #129: OTA upgrade not working

Korte68, 2020-04-27:
I tried to update the Nondemcu 2.3.4 to 2.3.5 but not updating.
offer the update, but do nothing.

heythisisnate, 2020-05-29:
Need more details here. Can you provide some output from the console?

Korte68, 2020-05-29:
Hi, I opened the Konnected app on my iPhone.
klick on the first device.
There is a blue big button:
A software update is available!
Tap here to update to software version 2.3.5

I tap there.
A pops window displayed:
Automatic Update
This will update...

I tap on the Ok.

A new popup window came up:

Automatic Update
The device is updating...

I tap the Ok.

That is all.

Device did not updating. stay on the previous version.

What do you mean on console output?

Any port should be opened on router?

Thank you

Nate Clark <notifications@github.com> ezt írta (időpont: 2020. máj. 29., P,
16:28):

heythisisnate, 2020-05-29:
Try wiping the device first by holding down the FLASH or RESET button for 10 seconds. Then try the OTA update again. After updated, you will have to re-sync the settings from your smart home platform.

Korte68, 2020-05-31:
Hi,
I tried it now. but still not working.

Nate Clark <notifications@github.com> ezt írta (időpont: 2020. máj. 29., P,
22:03):

closed 2020-10-13 by Korte68