konnected-io/konnected-security #132: Failure to save output state(s) on reboot

ceandraka, 2020-05-24:
Starting 2.3.4 the software was supposed to save output status, but I cannot get that to work. Using Home Assistant, but not sure that matters. Turn "on" output (pulls in a relay for me, actuating an LED) from HA. Connect to Konnected card via web browser. Click on "Restart Device" at bottom. After reboot, output is off.

Software version 2.3.4 and 2.3.5 tried, same results. Download form Github, install with Konnected flash tool. I do note that in both cases, firmware still shows as 2.3.0, even though firmware from github is installed labelled 2.3.4 and 2.3.5 in file name.

I suppose it is possible that HA is setting the output off upon reboot of Konnected card, but it should not. HA still shows output as on.

heythisisnate, 2020-05-25:
The expected behavior is that after a reboot, the device should retrieve the expected state from HA and re-set the state of the relay. So you'll see the switch turn off momentarily and then turn back on.

If you still believe it's not working as expected, please post the console output from the boot sequence.

ceandraka, 2020-05-25:
Nate:The output stays off after reboot, even though it was on prior.I will hook up my laptop later today and capture a reboot.Thanks for respondingChuck

ceandraka, 2020-05-25:
Nate:

 

I removed the board and plugged it into my desktop (easier to do esplorer). I booted the board after connecting by sending RTS twice. Then after it settled, I turned on the output port from HA (Yellow LED in HA). That turn on is seen in esplorer. I then caused a reboot by using the card’s web page and hitting “reset device”. I let it go until all it was getting is discovery hits. The “yellow LED” is still on in HA, but the output never turned on.

 

I am running software 2.3.5 on the device. This package includes firmware that shows up as 2.3.0. I am running HA 0.107.7 in a raspberry pi running OS 3.12

 

Here is the Esplorer window. Took multiple pastes due to garbage characters during reboot.

 

 

Konnected firmware 2.3.0                                                                                             (Initial boot up)

NodeMCU 2.2.0.0 build 20190805 powered by Lua 5.1.4 on SDK 2.2.1(6ab97e9)

Heap:    41712    Initializing Konnected (a4cf12df4b67)

Heap:    40456    Application Version:        2.3.5

Heap:    40192    Connecting to Wifi..

> Heap:                 41656    Wifi connected with IP: 192.168.1.55       Gateway:            192.168.1.2

Heap:    41840    Current Date/Time:         2020-05-25 15:55:03 UTC

Heap:    41760    Booting Konnected application

Heap:    41688    UPnP:   Listening for UPnP discovery

Heap:    41216    HTTP:    Starting server at http://192.168.1.55:22633

Heap:    41024    Loaded:                server

Heap:    38848    Initializing sensor pin:     1

Heap:    38816    Initializing sensor pin:     2

Heap:    38816    Initializing sensor pin:     5

Heap:    38816    Initializing sensor pin:     6

Heap:    38816    Initializing sensor pin:     7

Heap:    38816    Initializing sensor pin:     9

Heap:    38816    Initializing actuator pin: 8              on:         1              off:         0

Heap:    38240    REST Endpoint:  http://192.168.1.49:8123/api/konnected

Heap:    38208    Loaded:                application

 

Heap:    37288    HTTP Call:            200         state      0              pin          8

Heap:    37232    Initialized actuator Pin:  8              Trigger: 1              Initial state:        0

Heap:    38160    HTTP Call:            0              state      1              pin          1

Heap:    37528    HTTP Call:            200         state      1              pin          1

Heap:    38240    HTTP Call:            0              state      1              pin          2

[... 169 lines cut ...]
Heap:    33288    HTTP:    Discovery

Heap:    37992    Responding to UPnP Discovery request from 192.168.1.49:35182

Heap:    38160    Responding to UPnP Discovery request from 192.168.1.49:58166

Heap:    38176    Responding to UPnP Discovery request from 192.168.1.49:42349

Heap:    38216    Responding to UPnP Discovery request from 192.168.1.49:43813

Heap:    38200    Responding to UPnP Discovery request from 192.168.1.49:39402

Heap:    38200    Responding to UPnP Discovery request from 192.168.1.49:60125

Heap:    38216    Responding to UPnP Discovery request from 192.168.1.49:57591

 

PORT CLOSED   (No further action, a few minutes after reboot. Pin 8 output still appears off)

 

Thanks

Chuck

heythisisnate, 2020-05-25:
Thanks for the detail. I'll check into this.

ceandraka, 2020-05-27:
I re-ran the test, and captured the debug log on HA. The card in question as IP 55. Maybe this side will help as well> I see no mention of HA retransmitting the output pin.
[Log of HA on restart card 5.txt](https://github.com/konnected-io/konnected-security/files/4690898/Log.of.HA.on.restart.card.5.txt)

ceandraka, 2020-06-01:
As a temporary measure, I have made an automation that fires every minute. It checks the status of the output (in HA), and if on, it turns it on. A similar automation checks if it is off in HA, and turns it off. This means if the konnected card reboots, it will have the wrong output for no longer than a minute.

A rather brute force approach, and way too "chatty" for HA. However, it is effective until this is resolved

I think the automation to turn it off is not needed, because it reboots in an off state.

PR #141 (2020-09-29): fix comparison to ensure we restore state correctly
Comparing string to number was causing our state restoration logic to be bypassed.

fixes #132
closed 2020-09-29 by heythisisnate