konnected-io/konnected-security #135: No Boot issue when Auxiliary Devices Connected

blueknight768, 2020-06-22:
So there seems to be a odd issue with my konnected panel. I have confirmed this is a repeatable problem and has been happening much more recently. If I have all my door sensors and motion sensors connected as well as a smoke detector and the board boots up, connects to wifi I can successfully access the entities in home assistant and use everything as it should work. If in random occasion I would say about 70% of the time if home assistant reboots what happens with the konnected panel is it will dissapear from my ubiquity access point (continuous ping and connected devices list). The konnected panel at this point will be stuck indefinitely even when pushing the reboot button on the board or power cycling it. The only thing that fixes it is if I uplug the aux power devices from the board and reboot it, everything will reconnect as it should. This is all related in some fashion or another. Hopefully other people have this same problem as its rendering this system un-reliable. I work on my code sometimes remotely and have to restart HASS on occasion. This problem will come up and leave my home unprotected besides for cameras. 

The only things plugged into the AUX + Connections are a motion detector, a smoke alarm and a siren. Again when I unplug the + side and reboot the konnected v2 Panel it boots and connects fine. I cant imagine what could be the issue here. Maybe to much current draw or something? I even added one way diodes to the siren, motion and smoke alarm encase they were causing some kind of back feed, didn't help anything. 

When this first started happening I thought the firmware was becoming corrupted so I manually used konnected's re-flash tool with micro usb and latest 2.3.5 files from the github page. Except when this boot lockup I described earlier is in effect I cannot even flash the board it simply just says "connecting...___...___...___...___..." with audible ticking sounds with every three underscores coming from the board.

heythisisnate, 2020-06-23:
Couple things that come to mind that _could_ cause this issue:
Zone 6 and the ALARM/OUT zone (pins 9 and 8) must be _low_ on boot. Zone 6 cannot be used as an output for this reason, and ALARM/OUT must be a high-level trigger.

Post your configuration so we can rule out that as the issue.

blueknight768, 2020-12-22:
This was the issue. The last set of outputs I was using as a buzzer and was not low at boot. Kind of odd as I assume it's just mosfet based but when I removed the buzzer it boots fine.

closed 2022-08-23 by heythisisnate