konnected-io/konnected-security #148: ALARM1 e ALARM2 set to LOW when ON

errecla, 2021-06-17:
Hello. I'm working with Alarm PANEL PRO with 1.2.4 SW Version and 0.1.5-047de6 FW version and Hassio version core-2021.6.5. I need to work with ALARM1 and ALARM2 when on=LOW. My issue is that when I reboot hassio the switches ALARM1 and ALARM2 change status during reboot and siren ringing. Why?

heythisisnate, 2021-06-18:
When the device first boots, the outputs of ALARM1 and ALARM2 will momentarily be LOW until the software initializes. This cannot be avoided. It's recommended that you configure your alarm to be ON when the output is HIGH for this reason.

closed 2021-06-18 by heythisisnate

errecla, 2021-06-18:
Thanks for your answer but I have a siren that is singing when  cut OFF +12V. how can i configure ALARM1 and 2 output in LOW mode?
I attach the manual of my siren

[2manuale_ENG_SAEL2010_LED_ver_32.pdf](https://github.com/konnected-io/konnected-security/files/6678277/2manuale_ENG_SAEL2010_LED_ver_32.pdf)

heythisisnate, 2021-06-19:
@errecla You may need to use an external relay (controlled by OUT1 or OUT2) to make a normally-closed circuit with 12V+ to the siren input, then when activated it will break that circuit and activate the siren.