konnected-io/konnected-security #150: Board cannot reach Home Assistant. No zone update. Looping 10 times and reboots.

done7k, 2021-09-08:
3 boards were running quite  along time without an issue but then after some Hass update the board stopped updating statuses on Hass. Bellow is what I could get from one board
`Konnected firmware 2.3.0
NodeMCU 2.2.0.0 build 20190805 powered by Lua 5.1.4 on SDK 2.2.1(6ab97e9)
Heap: 	41712	Initializing Konnected (a9a9a9a9a9a9a9a9)
Heap: 	40424	Application Version: 	2.3.5
Heap: 	40176	Connecting to Wifi..
> Heap: 	41672	Wifi connected with IP: 	192.168.99.17	Gateway:	192.168.0.1
Heap: 	41832	Current Date/Time:	2021-09-07 21:04:35 UTC
Heap: 	41744	Booting Konnected application
Heap: 	41656	UPnP:	Listening for UPnP discovery
Heap: 	41168	HTTP: 	Starting server at http://192.168.99.17:22222
Heap: 	40984	Loaded: 	server
Heap:	38912	Initializing sensor pin:	1
Heap:	38880	Initializing sensor pin:	2
Heap:	38880	Initializing sensor pin:	5
Heap:	38880	Initializing sensor pin:	6
Heap:	38880	Initializing sensor pin:	7
Heap:	38880	Initializing actuator pin:	8	on:	1	off:	0
Heap:	38344	REST Endpoint:	https://192.168.99.99:8123/api/konnected
Heap: 	38312	Loaded: 	application
Heap:	37904	HTTP Call:	-1
Heap:	37872	Initialized actuator Pin:	8	Trigger:	1	Initial state:	0
Heap:	38504	HTTP Call:	0	state	1	pin	1
Heap:	37992	HTTP Call:	-1	state	1	pin	1
Heap:	38456	HTTP Call:	0	state	1	pin	1	retry	1
Heap:	37936	HTTP Call:	-1	state	1	pin	1	retry	1
Heap:	38456	HTTP Call:	0	state	1	pin	1	retry	2
Heap:	37944	HTTP Call:	-1	state	1	pin	1	retry	2
Heap:	38456	HTTP Call:	0	state	1	pin	1	retry	3
Heap:	37944	HTTP Call:	-1	state	1	pin	1	retry	3
Heap:	38456	HTTP Call:	0	state	1	pin	1	retry	4
Heap:	37928	HTTP Call:	-1	state	1	pin	1	retry	4
Heap:	38392	HTTP Call:	0	state	1	pin	1	retry	5
Heap:	37944	HTTP Call:	-1	state	1	pin	1	retry	5
Heap:	38456	HTTP Call:	0	state	1	pin	1	retry	6
Heap:	37944	HTTP Call:	-1	state	1	pin	1	retry	6
Heap:	38456	HTTP Call:	0	state	1	pin	1	retry	7
Heap:	37944	HTTP Call:	-1	state	1	pin	1	retry	7
Heap:	38456	HTTP Call:	0	state	1	pin	1	retry	8
Heap:	37944	HTTP Call:	-1	state	1	pin	1	retry	8
Heap:	38456	HTTP Call:	0	state	1	pin	1	retry	9
Heap:	37944	HTTP Call:	-1	state	1	pin	1	retry	9
Heap:	38456	HTTP Call:	0	state	1	pin	1	retry	10
Heap:	37936	HTTP Call:	-1	state	1	pin	1	retry	10
Heap:	37784	Retried 10 times and failed. Rebooting in 30 seconds.`

it seems it attempts to place a http.get() request to Hass but gets error -1. Unfortunately this error is not explained in NodeMCU doc here [https://nodemcu.readthedocs.io/en/release/modules/http/#httpget](url) , just a general error. At the same time in Hass' debug log there is no any msg for this event. Looks like the board does not even send this request. Any suggestion how this can be fixed?

done7k, 2021-09-09:
Seems the tiny esp8266 board does not work all time through SSL correctly. Switching to HTTP helps.

closed 2021-09-09 by done7k

heythisisnate, 2021-09-17:
We've had a few reports of this issue recently. Something may have changed in the LetsEncrypt certs that are commonly used for SSL/TLS in HA. The ESP8266 may not be powerful enough to handle all TLS certs.