konnected-io/konnected-security #151: Enhancement: Independently trigger output on state change

Bostwickenator, 2021-09-21:
From my reading of the documentation and code there doesn't seem to be a system to "ARM" the device. That is have the device take an action independently when a sensor state changes. Is this something that is explicitly not supported for philosophical reasons or just not implemented?

awulff73, 2021-09-21:
The logic for the alarm will be present in SmartHome platform you hook it up to.

Bostwickenator, 2021-09-21:
@awulff73 I don't think that addresses my request. If I'm mistaken can you please point me to the location in the code where the device will trigger an output without instruction from network device?

awulff73, 2021-09-21:
It's not in the code. Konnected is not made to work independently of a smarthome platform. That is where the logic for the alarm is.

Bostwickenator, 2021-09-21:
Cool, this looks like a valid feature request then!

awulff73, 2021-09-21:
I don't think you are understanding the use case. We won't be building alarm functionality into the code. We are not looking to produce a standalone alarm system. That's for the integration of the smarthome platform to handle.

Bostwickenator, 2021-09-21:
Just checking @awulff73 do you speak for @heythisisnate and the Konnected-IO team on this?

awulff73, 2021-09-21:
I work on the support side.

heythisisnate, 2021-09-21:
Andy is right. Konnected is not a standalone alarm system and we do not have plans to build arm/disarm logic or output triggering in the stock firmware. Philosophically, Konnected is a bridge for your wired devices to be monitored/controlled by one of our supported smart home platforms.
That said, we're committed to making our hardware open and extensible and customizable to your needs. I'd encourage you to look at running ESPHome on your Konnected device if you have a need for locally triggered actions. See: https://help.konnected.io/support/solutions/folders/32000036472

Bostwickenator, 2021-09-21:
Ah interesting I had assumed from the language on the product pages that at least minimal internal logic had to be provided if the system was being called a "retrofit" for existing alarm systems. As it stands a WiFi deauth attack would render the siren of a connected system inoperative. So it can't really hit a really major use case for traditional alarm systems. Having read through the code it seems extremely easy to add this functionality (which I'd happily contribute) but I don't suppose I should expect to convince you to change something that fundamental to your product offering. For my interest is (and please excuse my ignorance here) the Smart Things API fundamentally incompatible with the notion of a device which changes the state of the alarm siren autonomously?

heythisisnate, 2021-09-21:
It's not that the SmartThings API is "incompatible" per se. Konnected's design philosophy is to enable a smart home platform (SmartThings is just one option alongside Home Assistant, Hubitat, Alexa, etc) to act as the logic/brains of the alarm system so that you can leverage the power of home automation and all of your smart home devices in a security incident response. SmartThings in particular has the built-in SmartThings Home Monitor which manages the armed/disarmed/alert modes of the system, and Konnected simply relays sensor state and takes instructions from SmartThings to trigger actuators. This enables you do to things like:
* trigger z-wave/zigbee lights or sirens or door locks (or anything really) in response to an alarm
* consider inputs from a variety of sensors on multiple Konnected boards, as well as 3rd-party wireless sensors, to create an infinitely expandable alarm system
* automate arming and disarming based on presence detection, timers, or other automations entirely in the ST platform

Konnected is a "decentralized" system, which means that you could have multiple Konnected devices, in different parts of the house, all reporting into ST to create one single alarm system.

If you're concerned about WiFi deauth attacks, consider our Alarm Panel Pro which has Ethernet capability.

closed 2021-09-21 by heythisisnate

Bostwickenator, 2021-09-21:
Thanks for the info. It still seems to me that it would be possible to have some internal state determination without interfering from the usecases which you have specified. I'll roll something of my own. Thanks for including the PIN mappings in the documentation that will be useful.