konnected-io/konnected-security #155: Konnected Alarm Panel Reboots every 24 hours when a zone with Type: Switch is steady on for 24 hours

joshlarsen48, 2021-11-29:
When the state of a zone with type: Switch is in the ON state for 24 hours the Konnected Alarm Panel reboots at the 24 hour mark of that zone being ON. I have an old version and the new version of the boards and they both have the same issue. 

I have tested with both 2.3.5 and 3.0.1 software.

Preproduce by setting a zone type to Switch and then from SmartThings turn the state of that switch to ON and leave it that way and 24 hours later the board will reboot. The state of the switch is also lost and defaults back to OFF after the reboot.

heythisisnate, 2021-12-01:
Occasional reboots are normal and cannot be completely avoided. There are several reasons that the device may reboot from time to time, including wifi/network blips, re-authentication, power outages, etc. We can mitigate many of these, but not eliminate.

closed 2021-12-01 by heythisisnate

joshlarsen48, 2021-12-02:
Thanks Nate, The reboots was only a real issue because of the state not being remembered in issue https://github.com/konnected-io/konnected-security/issues/156. Since you fixed that the reboot is no longer a big issue.