konnected-io/konnected-security #156: State of Zone Type: Switch is lost at reboot

joshlarsen48, 2021-11-29:
If you have a zone with zone type of switch set to a state of ON and the board reboots that state is lost and the zone always returns to OFF after the reboot. This def becomes a problem if you have a state of ON on any switch zone for 24 or more hours as the board will auto reboot at that time. I opened https://github.com/konnected-io/konnected-security/issues/155 for that issue. 

Using SmartThings and I have tested with both 2.3.5 and 3.0.1 software.

Preproduce by setting a zone to zone type: Switch and then from SmartThings turn the state of that switch to ON and reboot the konnected device or leave it that way and 24 hours later the board will reboot on its own.

joshlarsen48, 2021-11-30:
I tested with state type of Siren and it also has the same issue and looses state every time the board reboots. 

A reboot happens every 24 hours of the state being on, during a power failure, when network connectivity is gone, ect.

joshlarsen48, 2021-11-30:
This is a pretty big issue when the claim is these are to be used for an alarm system. I keep being told by support the way Im using the board isn't normal use but the fact is the Siren could be on longer than 24 hours. Example below

If a user is gone for a weekend and their house was broken into lets say Friday night and they come home on Sun(More than 24 hours after the siren was turned on) they would normally expect the siren to be alarming when they got home saying someone broke in if someone actually did trigger the alarm. But with this bug the alarm would be silent since the board would have rebooted Sat night if not before because of the issue in issue #155. Lets hope the intruder wasn't there any longer when they came home because they would possibly have no idea till they were inside and found stuff missing. Might be a bit exaggerated and I also know you have a system that can work along side a real alarm system but you should get my drift or concern with the bug.

heythisisnate, 2021-11-30:
Agree that this is an issue that should be fixed. We may not be able to prevent all reboots, but at least the switch/siren state should revert to what it's supposed to be instead of defaulting back to OFF. Working on it.

heythisisnate, 2021-12-01:
Good news! This is now fixed via our cloud platform (no firmware update is necessary). Starting now, your switches/sirens/actuators should automatically resume their prior state following a reboot/re-connect. No action required to enable this.

closed 2021-12-01 by heythisisnate

joshlarsen48, 2021-12-02:
Thank you Nate. I had setup an alert for any state change and it turned off and right back on just a bit ago and sure enough The alarm panel rebooted so looks like your change def worked. Thanks for fixing this so quickly.