konnected-io/konnected-security #158: Unable to update firmware on 1st panel of Alarm V2 - keeps saying connecting

SeanHama, 2021-12-03:
[image]

I have the Alarm v2 system.

Steps
1: Connected and upgraded the v3.0.1 firmware on the 2nd panel (worked first time).
2: Connected laptop to the 1st panel and all it does it say connecting? Any ideas on how to upgrade this panel please?

awulff73, 2021-12-04:
It doesn't look like you have the Port which usually means you don't have a cord that can transmit data or you haven't installed the driver.

SeanHama, 2021-12-04:
The cable works fine and transmit data, I updated the 2nd alarm panel just fine. And have previously used the cable to upgrade the panel which is not connecting now.

The port is appear correctly, what alternatives
Or options do I have?

SeanHama, 2021-12-06:
Any ideas? My Konnected alarm is now not working correctly (the firmware is not the same on both boards) and the PIR sensors on the non-upgraded board not working.

awulff73, 2021-12-06:
Submit a ticket to help@konnected.io with a screenshot of your device status.

SeanHama, 2021-12-06:
> Submit a ticket to ***@***.*** with a screenshot of your device
> status.

done.

closed 2021-12-06 by heythisisnate