konnected-io/konnected-security #159: Firmware v3.0.1 no longer polling DS18B20 & reporting temp

andrewconnell, 2021-12-20:
REF: https://www.facebook.com/groups/konnected.io/posts/1556001921433830/

Verified with two different DS19B20 sensors & connected to two different pins. This was working with firmware 2.3.5. Once I upgraded to v3.x, the DS18B20 no longer reports temp.

Connected to the console using the web tool. In the console, I see it report the state of two pins & that it would poll for 3m (180000ms) on the pin the temp sensor is connected to, but it would never show.

heythisisnate, 2021-12-20:
So far have not been able to reproduce the problem. Two temp sensors connected on zone 1 working fine for me.
[image]
@andrewconnell can you share your complete zone configuration?

andrewconnell, 2021-12-20:
Hi @heythisisnate ... thanks for the reply. Sure... full zone setup:
- two reed switches (binary sensors) connected to zones 1 & 2 & working as expected
- temp sensor connected to zone 5 (also tried zone 6)

[image]
[image]

For good measure, here's my wiring setup:
[image]

- that's a 4K7 resistor between the signal (yellow) and VCC (red) connectors
- using the red alligator clip to power while debugging as my microUSB blocks plugging something into the 5V jumper

The only other thing that's strange is that when I configure it in HomeAssistant, I'm not seeing the polling interval change when I set it to something like 1 or 2 minutes... in the console, it's always showing 3m (180,000ms).

heythisisnate, 2022-08-23:
@h2zero could you take a look at this one and try to reproduce

andrewconnell, 2022-08-24:
FYI @heythisisnate, this has been working since v3.1 & also is working on the latest firmware. I should close it, but not sure if you want to leave it open to continue investigation..

heythisisnate, 2022-08-24:
Great. we will will close it then. I suspect it was a memory issue that has been resolved by the upgrade to the underlying firmware.

closed 2022-08-24 by heythisisnate