konnected-io/konnected-security #163: Wifi drops w/ reason code 15

mightymos, 2022-06-24:
Every few days my board drops wifi and fails to reconnect.
There seems to be no other issues with wifi devices in the house including surveillance cameras (on 2.4 ghz).
Router is RT3200 supporting 2.4ghz/5ghz and WPA2/WPA3.
I haven't changed anything recently or tried any debugging steps.

Serial debug gives:
```
brlr
Konnected firmware 3.0.0
NodeMCU 3.0.0.0 build 2021-11-22 17:54 powered by Lua 5.1.4 on SDK 3.0.1-dev(fce080e)
Heap:   39352   Initializing Konnected (f4cfa26ac36b)
Heap:   39224   Application Version:    3.0.1
Heap:   38944   Connecting to Wifi..
> Heap:         39368   Cannot connect to WiFi  Tribble Reason Code:    15
```

That apparently means:
https://nodemcu.readthedocs.io/en/release/modules/wifi/#wifieventmonreason

wifi.eventmon.reason.4WAY_HANDSHAKE_TIMEOUT 15

Why's it dropping and then failing to reconnect?

mightymos, 2022-06-25:
This is the closest help I've found:
https://github.com/esphome/issues/issues/3120

mightymos, 2022-06-28:
A router reboot allows reconnecting so I'll have to assume the problem is on that end.

mightymos, 2022-07-17:
Reopening because many other devices on home network seem to have no issues.

heythisisnate, 2022-07-17:
Sorry you're having trouble with this, but this isn't something that the Konnected team would address unless it's something reported by multiple users or specific to Konnected. The WiFi connection/reconnection is handled by the upstream nodemcu-firmware: https://github.com/nodemcu/nodemcu-firmware

closed 2022-07-17 by heythisisnate

mightymos, 2022-07-27:
Might be the last clue...ESP32 with tasmota seems to have no problems with wifi so far.
Works with detected reed sensor changes at doors.
Other tasmota based devices on the network (EZPlug, Sonoff bridge) seem to have no trouble.
[image]

heythisisnate, 2022-07-27:
I appreciate you following up on this issue, but again this is not something that will be addressed by the Konnected team. You may want to open an issue in the upstream firmware: https://github.com/nodemcu/nodemcu-firmware