konnected-io/konnected-security #170: OTA update failure on 3.1.1

sadahsan, 2022-08-22:
OTA update failure on 3.1.1
Tried few types to update through the konnected App but no luck.

heythisisnate, 2022-08-22:
what version are you updating from?

sadahsan, 2022-08-22:
App v3.0.1
Fw v3.0.0

Thanks

heythisisnate, 2022-08-22:
I was able to reproduce this and determined that we run into a memory consumption issue when attempting to do an OTA (LFS) update while the device is connected to AWS IoT.

_Potential solution:_ Shut down persistent AWS IoT websocket connection prior to beginning OTA download/update. @h2zero 

_Workaround:_ Reset the device first before performing the update to clear the AWS IoT connection. Hold down the RESET button for ~10 seconds when the device is idle until the blue light comes on solid, then release the button. The device will reboot. Perform the OTA update. Then go back to the Konnected App -> device -> Manage Zones -> Sync to Device to re-provision the device to connect to Konnected Cloud (AWS IoT).

sadahsan, 2022-08-22:
Thanks for looking into it so quickly, when do you expect to release the updated firmware with this fix?

h2zero, 2022-08-22:
I'll have a PR ready in the next few hours. It won't help you to update from your current version though, you'll likely need to follow the workaround method Nate described above.

sadahsan, 2022-08-22:
Ok thanks, will give it a go.

h2zero, 2022-08-23:
This may take a bit longer than expected. Closing the mqtt connection and disabling the associated timers is still not enough. This will require a more creative solution.

h2zero, 2022-08-23:
That's what I was thinking as well. I tried everything else so far but didn't have time to try the file/reboot yet.

PR #172 (2022-08-26): Fix OTA updates when connected to AWS.
This resolves an out of memory issue when updating via Over-The-Air while connected to a secure MQTT broker/AWS.
The process creates a file and writes the update info into it. The device is then rebooted and the file, if detected 
after connection to WiFi, is read from and the update performed using the data in the file.

Resolves #170
closed 2022-08-26 by heythisisnate